mantidproject/mantid #21334: Peak::findDetector now ~1000X slower

rosswhitfield, 2017-12-04:
Tested on RHEL7

Using CORELLI
```
cor = LoadEmptyInstrument(InstrumentName='CORELLI')
ws = CreatePeaksWorkspace(InstrumentWorkspace='cor')
p = ws.createPeak([1,1,1])
%timeit p.findDetector()
```
On `3.11.0` this is quick
```
%timeit p.findDetector()
1000 loops, best of 3: 793 µs per loop
```
But on `master`, `3.11.20171201.1727` as tested, this is very slow! About 1000X!
```
%timeit p.findDetector()
1 loops, best of 3: 551 ms per loop
```

Strangely for TOPAZ it now faster
```
topaz = LoadEmptyInstrument(InstrumentName='TOPAZ')
ws = CreatePeaksWorkspace(InstrumentWorkspace='topaz')
p = ws.createPeak([1,1,1])
%timeit p.findDetector()
```
On `3.11.0`:
```
%timeit p.findDetector()
1000 loops, best of 3: 407 µs per loop
```
On `master`:
```
%timeit p.findDetector()
1000 loops, best of 3: 215 µs per loop
```
I suspect this has to do CORELLI using tubes instead of, as for TOPAZ, rectangular detector.

This problem has carry on effects, I first observed this using `FindPeaksMD` which was running in about ~25 seconds with `3.11.0` but taking over 25 minutes with nightly.

Is this because of Instrument 2.0 changes as nothing has changed in `Peak`?

@OwenArnold @SimonHeybrock ?

OwenArnold, 2017-12-05:
This could be to do with the way that bounding box usage has been altered. Would be good to get a profiler check on where the time is being spent. 

Just a hunch, but you might find that avoiding the temporary `InstrumentRayTracer` in `Peak` [here](https://github.com/mantidproject/mantid/blob/6b6ffe7373fdd95c3e91fb60b10f8f1e283c77e1/Framework/DataObjects/src/Peak.cpp#L643) resolves the speed issues. `m_inst` is const so I think it would be very reasonable to create this once and cache it for future `findDetector` calls. For reference see #20721

rosswhitfield, 2017-12-05:
I don't think caching `InstrumentRayTracer` in `Peak` will help because typically you are not running `Peak::findDetector` lots of times on one peak but once on ~1000 different peaks.

rosswhitfield, 2017-12-05:
For a more complete example, grab [testmd.nxs](https://www.dropbox.com/s/pquxzntcmzhp6id/testmd.nxs?dl=0) (1.2Gb), then run:
```
md=LoadMD('testmd.nxs')
FindPeaksMD(InputWorkspace=md,DensityThresholdFactor=50000, OutputWorkspace='peaks')
```
On `3.11.0`:
```
FindPeaksMD(InputWorkspace=md,DensityThresholdFactor=50000, OutputWorkspace='peaks')
FindPeaksMD-[Notice] FindPeaksMD started
FindPeaksMD-[Notice] Number of peaks found: 1549
FindPeaksMD-[Notice] FindPeaksMD successful, Duration 27.21 seconds
```
and with nightly:
```
FindPeaksMD(InputWorkspace=md,DensityThresholdFactor=50000, OutputWorkspace='peaks')
FindPeaksMD-[Notice] FindPeaksMD started
FindPeaksMD-[Notice] Number of peaks found: 1549
FindPeaksMD-[Notice] FindPeaksMD successful, Duration 29 minutes 18.00 seconds
```
27 seconds to 29 minutes!

rosswhitfield, 2017-12-06:
Reverting #20850, #20721, #21267, #21146 and #20274 _fixes_ the problem. I have a [branch](https://github.com/rosswhitfield/mantid/tree/revert_bounding_box) ready to revert the changes if this can't be fixed before the next release.

OwenArnold, 2017-12-13:
@rosswhitfield sorry I've been too busy to do anything about this over the last week. I'll take a look at this and try to establish how it can be fixed.

rosswhitfield, 2017-12-13:
Thanks @OwenArnold

PR #21471 (2018-01-04): FindPeaksMD performance issue for tube based instruments
Fixes #21334. Gets the performance comparable to what it was previously.

Special handling for tubes and banks of tubes.

**Tester**
- Code Review
- Performance Testing using issue #21334
closed 2018-01-04 by AndreiSavici