mantidproject/mantid #21360: `mantid.plots` not included in Framework

rosswhitfield, 2017-12-05:
If you compile just the Framework, nothing else, mantid will fail to import into python

```
$ bin/mantidpython 
Python 3.6.3 (default, Oct 24 2017, 14:48:20) 
Type 'copyright', 'credits' or 'license' for more information
IPython 6.2.1 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import mantid
FrameworkManager-[Notice] Welcome to Mantid 3.11.20171205.1322
FrameworkManager-[Notice] Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-1-002efd284d2a> in <module>()
----> 1 import mantid

~/mantid/Release/temp/bin/mantid/__init__.py in <module>()
     76 import mantid.dataobjects as dataobjects
     77 import mantid._plugins as _plugins
---> 78 import mantid.plots as plots
     79 
     80 ###############################################################################

ModuleNotFoundError: No module named 'mantid.plots'
```

peterfpeterson, 2017-12-07:
Fortunately, this doesn't affect the packages
```
$ rpm -qilp ./mantidnightly-3.11.20171206.2201-1.el7.x86_64.rpm  | grep mantid/plots
/opt/mantidnightly/bin/mantid/plots
/opt/mantidnightly/bin/mantid/plots/__init__.py
/opt/mantidnightly/bin/mantid/plots/__init__.pyc
/opt/mantidnightly/bin/mantid/plots/_functions.py
/opt/mantidnightly/bin/mantid/plots/_functions.pyc
/opt/mantidnightly/share/doc/api/python/mantid/plots
/opt/mantidnightly/share/doc/api/python/mantid/plots/index.html
```

rosswhitfield, 2018-02-07:
This is also causing our [pylint build to fail](http://builds.mantidproject.org/job/master_pylint/2782/console) so it should be fixed.

peterfpeterson, 2018-02-08:
```shell
cmake -G Ninja ~/code/mantid -DENABLE_MANTIDPLOT=False -DENABLE_WORKBENCH=False
ninja
bin/mantidpython
```
followed by `import mantid` works for me already.

rosswhitfield, 2018-02-08:
This fails if you don't set `-DENABLE_MANTIDPLOT=False` but just do `ninja Framework`

commit 1bf6fb4f (2018-02-08): Add dependency on PythonInterface

This fixes #21360
PR #21721 (2018-02-09): Add dependency on PythonInterface
**To test:**

```shell
cmake -G Ninja ~/code/mantid -DENABLE_MANTIDPLOT=False -DENABLE_WORKBENCH=False
ninja
bin/mantidpython
```
followed by
```python
import mantid
```

This fixes #21360

*Does not need to be in the release notes* because it is a fix for framework builds with mantidplot enabled.

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the code conform to the [coding standards](http:// [... cut]
closed 2018-02-09 by martyngigg