mantidproject/mantid #21379: LoadEmptyInstrument produces workspace with nonzero Y and E values

MikeHart85, 2017-12-07:
### Expected behavior

LoadEmptyInstrument should produce an empty workspace, with the specified instrument loaded.

### Actual behavior

The workspace produced by LoadEmptyInstrument contains nonzero Y values and Error values.

Monitors have a value of `2`, others a value of `1` (both Y and E).

This does no happen when using the WorkspaceFactory to `create` a workspace. That produces a workspace where all Y and E values are `0` as expected.

### Steps to reproduce the behavior

```python
In [1]: ws = LoadEmptyInstrument(InstrumentName='SANS2D')
In [2]: ws.readY(123)
Out[2]: array([ 1.])
In [3]: ws.readE(123)
Out[3]: array([ 1.])

In [4]: ws2 = WorkspaceFactory.create(ws)
In [5]: ws2.readY(123)
Out[5]: array([ 0.])
In [6]: ws2.readE(123)
Out[6]: array([ 0.])
```
Double click the workspace to see all of them.

soininen, 2017-12-07:
Note that there are *DetectorValue* and *MonitorValue* properties to control the values the workspace is filled with.

MikeHart85, 2017-12-07:
Thanks, I hadn't seen those.

In that case, I wonder why they have non-zero defaults. The descriptions ...

> This value **affects the colour** of the detectors in the instrument display window (default 1)
> This value **affects the colour** of the monitors in the instrument display window (default 2)

... seem to imply this was done to make detectors and monitors show up in different colours in the Instrument View (despite there being no real data).

However, when I go to "Show Instrument" on a workspace loaded by `LoadEmptyWorkspace`, everything shows up white anyway.

Perhaps that could be fixed... though I would say the defaults should still be `0`. It's quite surprising to find data you didn't explicitly ask for in a supposedly empty instrument workspace. Even if the colours worked as intended, this doesn't seem like a good enough reason to inject false Y and E values by default.

soininen, 2017-12-07:
The problem is, there may be scripts out there where the default values (1 for detectors, 2 for monitors) are actually relied upon. Changing them to zero will break the Python API.

closed 2018-06-28 by NickDraper