mantidproject/mantid #21380: WorkspaceFactory creates workspaces with incorrect bin boundaries

MikeHart85, 2017-12-07:
### Expected behavior

Unless otherwise specified, it makes most sense for bin boundaries to start at `0`. 

When creating a workspace based on another, the bin boundaries would likely be expected to match.

### Actual behavior

`WorkspaceFactory.create` produces workspaces with bin boundaries that begin at `1`.

This happens both when creating a workspace by class name, and when passing in another workspace to use as a "parent" (disregarding the bin boundaries of the parent).

This does not happen when creating a workspace with `LoadEmptyInstrument`.

### Steps to reproduce the behavior

```python
In [1]: ws1 = WorkspaceFactory.create("Workspace2D", 122888, 2, 1)
In [2]: ws1.readX(123)
Out[2]: array([ 1.,  2.])

In [3]: ws2 = LoadEmptyInstrument(InstrumentName="SANS2D")
In [4]: ws2.readX(123)
Out[4]: array([ 0.,  1.])

In [5]: ws3 = WorkspaceFactory.create(ws2)
In [6]: ws3.readX(123)
Out[6]: array([ 1.,  2.])
```

SimonHeybrock, 2017-12-13:
Not a fix, but I recommend using the newer factory functions provided in `MantidDataObjects/WorkspaceCreation.h`. I believe they exhibit a more sensible behavior, and also allow for direct specification of the desired bin edges.

Disclaimer: I wrote them myself.

NickDraper, 2018-06-28:
The Bin boundaries in this case should always be overwritten by the caller, as you have created an empty workspace, I can't think of a single case where any default would be correct.

closed 2018-06-28 by NickDraper