mantidproject/mantid #21444: MaxEnt extend so chi and MaxAngle outputted so can get max iter per spectrum

Anders-Markvardsen, 2017-12-15:
The work in #21435 extends maxent to output number of evolution once converged, it adjust e.g. MaxAngle ouptut for all spectra to be the max number found among all spectra. 

In this issue:

* adjust MaxAngle and chi workspaces so that for each spectrum to have length equal to max maxent iterations for that spectrum
* update the MaxEnt doc (rst file) this information that this is done

AnthonyLim23, 2017-12-19:
can we talk about this in the new year? I don't understand what you are asking for.

Anders-Markvardsen, 2018-01-05:
I will make a start on this one

KarlPalmen, 2018-03-14:
@Anders-Markvardsen  I had a look at the workspaces using the script modified so that the workspaces created by the second P run were named differently (`...P2`), so I could examine all workspaces. I find that `evolAngleC` workspace not only has different length spectra to the `evolAngleP` and `evolAngleC` the values appear to be different. 

Do you want the first spectrum of `evolAngleC` to equal `evolAngleP` and the second spectrum of `evolAngleC` to equal `evolAngleP2`?

I'll examine the code to see how this could be ensured.

KarlPalmen, 2018-03-15:
@Anders-Markvardsen  After lots of debugging and examining workspaces, I found out that the script combined the two workspaces incorrectly. The error values were in the wrong order. After correcting to 
```
CreateWorkspace(OutputWorkspace='wsCombined',DataX=X+X2+X+X2,DataY=YRe+YRe2+YIm+YIm2,DataE=E+E2+E+E2,NSpec=4)
```
I get the results:
```
Number of iterations dataset1: 11
Number of iterations dataset2: 7
Number of iterations dataset1: 11
Number of iterations dataset2: 11
```
The dataset2 of the combined workspace was padded out with zeroes. All behaviour seems to the OK, except that the values of dataset2 in the combined workspace are different for the 7 non-zero values and start with the same value as for the dataset1. These datasets are for the angle, the Chi values are the same for separate and combined workspaces, except for the zero padding.

KarlPalmen, 2018-03-15:
@Anders-Markvardsen  A further check of the angles show they are identical for the separate and combined workspaces, except for the zero padding.

I no longer see an issue here, expect possibly removing the zero padding.

KarlPalmen, 2018-03-15:
To ensure error-free combining of workspaces, one could modify the algorithm `AppendSpectra` to deal with complex valued spectra as in this issue. In the current form, it won't do this, because we use `(R1, R2, ..., I1,  I2, ...)` ordering rather than `(R1, I1, R2, I2,...)` ordering. http://docs.mantidproject.org/nightly/algorithms/AppendSpectra-v1.html

KarlPalmen, 2018-03-15:
I checked the documentation at http://docs.mantidproject.org/nightly/algorithms/MaxEnt-v1.html and find that its descriptions of the number of bins in `evolAngle` and `evolChi` are not correct. It is the maximum number of iterations DONE (11) rather than the maximum number of iterations ALLOWED (20,000).

Anders-Markvardsen, 2018-03-15:
Many thanks for this investigations.

In terms of this issue I would like the output to be:

```
Number of iterations dataset1: 11
Number of iterations dataset2: 7
Number of iterations dataset1: 11
Number of iterations dataset2: 7
```

I am not in tomorrow. @AnthonyLim23 will also be able to help with question of this issue.

I agree that the document is out-of-date for evolAngle and evalChi. Please update these thanks

AnthonyLim23, 2018-03-16:
Here is the code I changed to trim the zeros (but kept a ws with all spectra the same length) https://github.com/mantidproject/mantid/pull/21435/files

KarlPalmen, 2018-03-16:
I found that @AnthonyLim23 attempted to limit the spectra of the `evolAngle` and `evolChi` to the actual number of iterations done rather then the maximum number of iterations done, so producing a ragged workspace, but this fell foul of poor support of ragged workspaces then. The code for this is at  https://github.com/mantidproject/mantid/pull/21435/files .

KarlPalmen, 2018-03-16:
Actually the code in the link is the actual code now. Ragged workspace was not attempted.

Anders-Markvardsen, 2018-03-21:
@KarlPalmen let e.g. @AnthonyLim23 or me know when you have a PR reading for testing for this issue

PR #22163 (2018-04-04): Maxent algorithm output of iteration records and use of complex data
Ensure that Maxent algorithm returns for each spectrum, in the two workspaces that record the iterations exactly one element per iteration with no padding with zeros. Update the documentation and provide a usage example covering this and also how to combine two workspaces with complex data.

**To test:**

Build the documentation check that it is OK and up to date.

Run the usage example added to the documentation. 
Then select workspaces `EvolAngleP` & `EvolAngleP2` and plot them, then plot `EvolAngleC` both plots should be i [... cut]
closed 2018-04-04 by martyngigg