mantidproject/mantid #21466: Materials without chemical formulas?

soininen, 2017-12-20:
In some experiments, the investigated materials are so complex that the exact chemical formula is unwieldy or simply unknown. However, this shouldn't prevent one from using [`SetSample`](http://docs.mantidproject.org/nightly/algorithms/SetSample-v1.html) to set the scattering cross sections of the material. Using a fake formula would work, but it is not ideal. It would be better if the chemical formula was optional in which case the all cross sections need to be specified.

SpencerHowells, 2017-12-20:
@soininen Have you tried SetSampleMaterial? We had an option for inputting cross-sections or formula and removed the former. Perhaps we should put it back in!

soininen, 2017-12-20:
@SpencerHowells `SetSample` redirects the material properties to `SetSampleMaterial`. It seems that the *ChemicalFormula* property is not mandatory in `SetSampleMaterial`, I fact seem to have overlooked. I'll need to test if things actually work already without the formula.

soininen, 2017-12-20:
The *ChemicalFormula* property in `SetSampleMaterial` seems to indeed be mandatory. Also, the `Material` class in Mantid cannot be constructed without the formula.

closed 2019-01-28 by martyngigg