mantidproject/mantid #21467: Confusing explanation of isotopes in SetSampleMaterial documentation

soininen, 2017-12-20:
There seems to be some confusion among users in [`SetSampleMaterial`](http://docs.mantidproject.org/nightly/algorithms/SetSampleMaterial-v1.html#algm-setsamplematerial-v1) docs about the isotope syntax.

SpencerHowells, 2017-12-20:
@soininen It's just within the brackets eg (Li7)

soininen, 2017-12-20:
@SpencerHowells I gathered as much, but this is not clearly stated in the documentation. In fact, the Li7 example there is without the parentheses.

SpencerHowells, 2017-12-20:
@soininen The info in the wiki for SetSampleMaterial has an example for a molecular formula which clearly states (Li7)

soininen, 2018-01-10:
@SpencerHowells Just before the example the documentation states "(as in Li7)" without the parentheses.