mantidproject/mantid #21468: SetSampleMaterial: no possibility to set number density for formula units

soininen, 2017-12-20:
Internally Mantid uses atom number densities. This is also reflected in [`SetSampleMaterial`](http://docs.mantidproject.org/nightly/algorithms/SetSampleMaterial-v1.html) where the input property *SampleNumberDensity* expects the density per atom. In some disciplines, however, the use number densities per formula unit would be more convenient. It should be investigated if an additional input property for the formula unit density could be added to `SetSampleMaterial` and the density converted to the Mantid convention using the *ChemicalFormula* information within the algorithm.

SpencerHowells, 2017-12-20:
@soininen Our GUIs already use the option of mass density or number density. See for example CalculateMonteCarloAbsorption.py

SpencerHowells, 2017-12-20:
@soininen The whole Mantid community had a long discussion on number density. This included Ian Bush & Miguel. The decision was that it is atoms/A^3. So if you use mass density and formula you get that value.

soininen, 2018-01-10:
@SpencerHowells I'm perfectly fine with the internal representation of atoms/Å^3. The idea is just to give users the opportunity to give the quantity in units of formula units/Å^3 if they so wish. This can be converted to whatever internal representation Mantid uses. Currently, users need to calculate atoms/Å^3 manually which is error prone. This could be done automatically within the algorithm based on given formula and the formula units/Å^3 value.

SpencerHowells, 2018-01-10:
@soininen We have done it specifically to avoid user error. The input would be the chemical formula and the mass density (the measured unit). The algo calculates the cross-sections, number of atoms and then the number density. Isn't this what they are doing manualy?

SpencerHowells, 2019-02-12:
@soininen Shall we change our interfaces to give this option?

soininen, 2019-02-13:
@SpencerHowells This was requested by users who - to my knowledge, at least - don't use the indirect geometry interfaces. I cannot see why this option wouldn't be useful there as well, but at least from our part it is nothing urgent.

PR #24752 (2019-03-04): Add options to specify number densities as formula units / Ångström^3
Internally we use atoms / Ångström^3 but it is beneficial to permit users to give the number densities as formula units / Ångström^3. This is useful when the chemical formula is complex. Mantid can easily calculate the actual number of atoms and convert to the internal number density representation.

This PR enables the new input unit options in `mantid.kernel.MaterialBuilder`, `SetSampleMaterial` and `LoadSampleEnvironment`.

An additional cleanup considers the return type of some `Material` returning functions which were r [... cut]
closed 2019-03-04 by NickDraper