mantidproject/mantid #21505: FindFilesThreadPoolManagerTest Fails on Fedora 26

quantumsteve, 2018-01-04:
### Expected behavior

All test pass on Fedora

### Actual behavior
```
[svh@pc102795 bin]$ gdb ./MantidQtWidgetsCommonTestQt4
GNU gdb (GDB) Fedora 8.0.1-33.fc26
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-redhat-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./MantidQtWidgetsCommonTestQt4...done.
(gdb) run FindFilesThreadPoolManagerTest 
Starting program: /home/svh/Documents/MantidProject/build/bin/MantidQtWidgetsCommonTestQt4 FindFilesThreadPoolManagerTest
Missing separate debuginfos, use: dnf debuginfo-install glibc-2.25-12.fc26.x86_64
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
Running 2 tests[New Thread 0x7fffce930700 (LWP 13351)]
[Thread 0x7fffce930700 (LWP 13351) exited]
.
Thread 1 "MantidQtWidgets" received signal SIGSEGV, Segmentation fault.
0x00007fffd3387ac1 in gtk_container_add () from /lib64/libgtk-x11-2.0.so.0
Missing separate debuginfos, use: dnf debuginfo-install GConf2-3.2.6-17.fc26.x86_64 OCE-foundation-0.18.1-1.fc26.x86_64 OCE-modeling-0.18.1-1.fc26.x86_64 PackageKit-gtk3-module-1.1.6-2.fc26.x86_64 adwaita-gtk2-theme-3.22.3-2.fc26.x86_64 atk-2.24.0-1.fc26.x86_64 avahi-glib-0.6.32-7.fc26.x86_64 avahi-libs-0.6.32-7.fc26.x86_64 boost-date-time-1.63.0-10.fc26.x86_64 boost-filesystem-1.63.0-10.fc26.x86_64 boost-regex-1.63.0-10.fc26.x86_64 boost-serialization-1.63.0-10.fc26.x86_64 boost-system-1.63.0-10.fc26.x86_64 bzip2-libs-1.0.6-22.fc26.x86_64 cairo-1.14.10-1.fc26.x86_64 compat-openssl10-1.0.2m-1.fc26.x86_64 dbus-glib-0.108-2.fc26.x86_64 dbus-libs-1.11.18-1.fc26.x86_64 expat-2.2.4-1.fc26.x86_64 fontconfig-2.12.6-4.fc26.x86_64 freetype-2.7.1-9.fc26.x86_64 gdk-pixbuf2-2.36.9-1.fc26.x86_64 glib2-2.52.3-2.fc26.x86_64 gnome-vfs2-2.24.4-22.fc26.x86_64 graphite2-1.3.10-1.fc26.x86_64 gsl-2.3-1.fc26.x86_64 gstreamer1-1.12.3-1.fc26.x86_64 gstreamer1-plugins-base-1.12.3-1.fc26.x86_64 gtk2-2.24.31-4.fc26.x86_64 harfbuzz-1.4.4-1.fc26.x86_64 hdf5-1.8.18-5.fc26.x86_64 jsoncpp-1.8.1-3.fc26.x86_64 libICE-1.0.9-9.fc26.x86_64 libSM-1.2.2-5.fc26.x86_64 libX11-1.6.5-2.fc26.x86_64 libXau-1.0.8-7.fc26.x86_64 libXcomposite-0.4.4-9.fc26.x86_64 libXcursor-1.1.14-8.fc26.x86_64 libXdamage-1.1.4-9.fc26.x86_64 libXext-1.3.3-5.fc26.x86_64 libXfixes-5.0.3-2.fc26.x86_64 libXi-1.7.9-2.fc26.x86_64 libXinerama-1.1.3-7.fc26.x86_64 libXrandr-1.5.1-2.fc26.x86_64 libXrender-0.9.10-2.fc26.x86_64 libblkid-2.30.2-1.fc26.x86_64 libcanberra-0.30-13.fc26.x86_64 libcanberra-gtk2-0.30-13.fc26.x86_64 libcap-2.25-5.fc26.x86_64 libcrypt-nss-2.25-12.fc26.x86_64 libdatrie-0.2.9-4.fc26.x86_64 libffi-3.1-12.fc26.x86_64 libgcc-7.2.1-2.fc26.x86_64 libgcrypt-1.7.9-1.fc26.x86_64 libglvnd-1.0.0-1.fc26.x86_64 libglvnd-egl-1.0.0-1.fc26.x86_64 libglvnd-glx-1.0.0-1.fc26.x86_64 libgomp-7.2.1-2.fc26.x86_64 libgpg-error-1.25-2.fc26.x86_64 libicu-57.1-7.fc26.x86_64 libjpeg-turbo-1.5.3-1.fc26.x86_64 libmount-2.30.2-1.fc26.x86_64 libogg-1.3.2-6.fc26.x86_64 libpng-1.6.28-2.fc26.x86_64 libselinux-2.6-7.fc26.x86_64 libstdc++-7.2.1-2.fc26.x86_64 libtdb-1.3.13-1.fc26.x86_64 libthai-0.1.25-2.fc26.x86_64 libtool-ltdl-2.4.6-17.fc26.x86_64 libuuid-2.30.2-1.fc26.x86_64 libvorbis-1.3.5-2.fc26.x86_64 libwebp-0.6.1-1.fc26.x86_64 libxcb-1.12-3.fc26.x86_64 libxml2-2.9.4-2.fc26.x86_64 libxslt-1.1.29-1.fc26.x86_64 lz4-libs-1.8.0-1.fc26.x86_64 mesa-libGLU-9.0.0-11.fc26.x86_64 muParser-2.2.5-3.fc26.x86_64 mxml-2.9-2.fc26.x86_64 nss-softokn-freebl-3.34.0-1.0.fc26.x86_64 openssl-libs-1.1.0g-1.fc26.x86_64 orc-0.4.27-1.fc26.x86_64 pango-1.40.12-1.fc26.x86_64 pcre-8.41-3.fc26.x86_64 pixman-0.34.0-3.fc26.x86_64 poco-debug-1.7.8p2-3.fc26.x86_64 proj-4.9.3-2.fc26.x86_64 python2-libs-2.7.14-4.fc26.x86_64 qscintilla-2.10-5.fc26.x86_64 qt-4.8.7-28.fc26.x86_64 qt-mobility-location-1.2.2-0.26.20140317git169da60c.fc26.x86_64 qt-mobility-sensors-1.2.2-0.26.20140317git169da60c.fc26.x86_64 qt-x11-4.8.7-28.fc26.x86_64 qtwebkit-2.3.4-16.fc26.x86_64 sqlite-libs-3.20.1-1.fc26.x86_64 sssd-client-1.16.0-4.fc26.x86_64 systemd-libs-233-7.fc26.x86_64 tbb-2017.7-1.fc26.x86_64 xz-libs-5.2.3-2.fc26.x86_64 zlib-1.2.11-2.fc26.x86_64
(gdb) bt
#0  0x00007fffd3387ac1 in gtk_container_add () from /lib64/libgtk-x11-2.0.so.0
#1  0x00007ffff763bade in QGtkStylePrivate::setupGtkWidget(_GtkWidget*) () from /lib64/libQtGui.so.4
#2  0x00007ffff763bfb9 in QGtkStylePrivate::addWidget(_GtkWidget*) [clone .part.35] () from /lib64/libQtGui.so.4
#3  0x00007ffff763c206 in QGtkStylePrivate::initGtkWidgets() const () from /lib64/libQtGui.so.4
#4  0x00007ffff7622be1 in QGtkStyle::QGtkStyle() () from /lib64/libQtGui.so.4
#5  0x00007ffff75ac4d8 in QStyleFactory::create(QString const&) () from /lib64/libQtGui.so.4
#6  0x00007ffff72a9d88 in QApplication::style() () from /lib64/libQtGui.so.4
#7  0x00007ffff72aa12d in QApplicationPrivate::initialize() () from /lib64/libQtGui.so.4
#8  0x00007ffff72aa24e in QApplicationPrivate::construct(_XDisplay*, unsigned long, unsigned long) () from /lib64/libQtGui.so.4
#9  0x00007ffff72aa4ef in QApplication::QApplication(int&, char**, int) () from /lib64/libQtGui.so.4
#10 0x00000000005e57f0 in FindFilesThreadPoolManagerTest::test_starting_new_search_cancels_currently_running_search (this=0xacb390)
    at /home/svh/D
ocuments/MantidProject/mantid/qt/widgets/common/test/FindFilesThreadPoolManagerTest.h:75
#11 0x00000000005e5db3 in TestDescription_suite_FindFilesThreadPoolManagerTest_test_starting_new_search_cancels_currently_running_search::runTest (
    this=0x992160 <testDescription_suite_FindFilesThreadPoolManagerTest_test_starting_new_search_cancels_currently_running_search>) at qt/widgets/common/qt4/FindFilesThreadPoolManagerTest.cpp:31
#12 0x00000000005ec36d in CxxTest::RealTestDescription::run (this=0x992160 <testDescription_suite_FindFilesThreadPoolManagerTest_test_starting_new_search_cancels_currently_running_search>)
    at /home/svh/Documents/MantidProject/mantid/Testing/Tools/cxxtest/cxxtest/RealDescriptions.cpp:96
#13 0x00000000005ef7d3 in CxxTest::TestRunner::runTest (this=0x7fffffffd6af, td=...) at /home/svh/Documents/MantidProject/mantid/Testing/Tools/cxxtest/cxxtest/TestRunner.h:94
#14 0x00000000005ef6cc in CxxTest::TestRunner::runSuite (this=0x7fffffffd6af, sd=...) at /home/svh/Documents/MantidProject/mantid/Testing/Tools/cxxtest/cxxtest/TestRunner.h:79
#15 0x00000000005ef576 in CxxTest::TestRunner::runWorld (this=0x7fffffffd6af) at /home/svh/Documents/MantidProject/mantid/Testing/Tools/cxxtest/cxxtest/TestRunner.h:63
#16 0x00000000005ef378 in CxxTest::TestRunner::runAllTests (listener=...) at /home/svh/Documents/MantidProject/mantid/Testing/Tools/cxxtest/cxxtest/TestRunner.h:24
#17 0x00000000005f5258 in CxxTest::XUnitPrinter::run (this=0x7fffffffd770) at /home/svh/Documents/MantidProject/mantid/Testing/Tools/cxxtest/cxxtest/XUnitPrinter.h:30
#18 0x00000000005f6021 in CxxTest::Main<CxxTest::XUnitPrinter> (tmp=..., argc=2, argv=0x7fffffffdd78) at /home/svh/Documents/MantidProject/mantid/Testing/Tools/cxxtest/cxxtest/TestMain.h:106
#19 0x00000000005eb467 in main (argc=2, argv=0x7fffffffdd78) at qt/widgets/common/qt4/MantidQtWidgetsCommonTestQt4_runner.cpp:27
(gdb)
```

### Steps to reproduce the behavior

ctest -R MantidQtWidgetsCommonTestQt4_FindFilesThreadPoolManagerTest

### Platforms affected

Fedora 26

quantumsteve, 2018-01-04:
This test was added in #20954. @samueljackson92 any idea what's going on?

samueljackson92, 2018-01-05:
@quantumsteve It looks like they're both falling over in the `QApplication` constructor. Both this test and the one in #21504 create a `QApplication` before running because they need an event loop for processing queued events between threads. Creating a `QApplication` object seemed like the simplest way to get one in the test at the time of writing. 

Why this is happening on Fedora 26 and not on any of the other builds I do not know.

quantumsteve, 2018-01-05:
Interestingly, the test `MantidQtWidgetsCommonTestQt5_FindFilesThreadPoolManagerTest` passed.

quantumsteve, 2018-12-20:
Fedora 26 no longer supported. Closing issue.

closed 2018-12-20 by quantumsteve