mantidproject/mantid #21553: extractShape or similar for OFF style Geometries

OwenArnold, 2018-01-15:
We would require something like this..
``Object shape = extractShape(const H5Group& group)``; 

Extensive unit testing would be required.

Please speak to @OwenArnold and @matthew-d-jones when you are closer to the point of doing this. @matthew-d-jones will be able to provide a nexus file with OFF shapes provided. This would probably be best done as part of the `NexusGeometry` library currently under development [here](https://github.com/mantidproject/mantid/pull/21369)

matthew-d-jones, 2018-01-15:
I can provide more information and other examples when you get to this, but here is a simple, example NeXus file containing an icosahedral sample:
https://www.dropbox.com/s/l0vywkf3ujnptyw/icosahedron_sample_example.nxs?dl=0
[image]

KarlPalmen, 2018-03-09:
I note that the OFF format in the Nexus file provided by @matthew-d-jones differs from the text format described in Wikipedia. It has 3 lists: 
* A list of vertices
* A list of vertex indices referring to previous list
* A list of indices referring to previous list giving start of each face in that list

OwenArnold, 2018-03-13:
@KarlPalmen regarding the question you left on #21369. The nexus geometry parsing code is not yet ready to be merged into master, however if you wish to branch from it, you can go-ahead and do so. Note that `NexusGeometryParser` is designed to create a complete instrument geometry. That being said, you should have a look at `NexusGeometryParser::parseNexusShape` as this will read the H5 Group and return an IObject for the shape.

KarlPalmen, 2018-03-13:
This would require `NexusGeometryParser::parseNexusShape` to be added to `NexusGeometryParser.h` alongside `NexusGeometryParser::createInstrument`?

OwenArnold, 2018-03-13:
@KarlPalmen unless you are planning on starting this immediately. I would like to think a bit longer about how to sensibly arrange the Nexus Geometry code. What are your timescales? When do you need this by?

KarlPalmen, 2018-03-13:
Timescale is about 1 week. I'm waiting for this issue here to be reviewed and would respond to such a review first.

OwenArnold, 2018-03-13:
@KarlPalmen I've made some quick changes to get you going. See https://github.com/ScreamingUdder/mantid/blob/c5f4a11251eb8b7bf2761809b81d12952d85c17e/Framework/NexusGeometry/inc/MantidNexusGeometry/NexusShapeFactory.h#L56-L59 

You would still need to work off my branch.

KarlPalmen, 2018-03-13:
This work requires the branch of #12828 merging into its base branch, because it is an extension of the `LoadShape` algorithm, which has been created there.

KarlPalmen, 2018-03-14:
Once such merger has occurred, I'll add a new function `readOFFshape` to `LoadShape` that takes the Nexus file pointer as argument and goes to the sample, opens it, then to the shape opens it and reads the three data blocks, then after checking them calls `nexusShapeFactory.createFromOFFMesh` with the these 3 data blocks in the arguments.

OwenArnold, 2018-03-14:
@KarlPalmen I would check with @Anders-Markvardsen what his intention is with this issue, as he has authored the description. I may be that this `LoadShape` algorithm is required to read a single shape specified in a single OFF file and apply it somewhere to an existing component in an existing instrument. In other words, Anders might intend that this algorithm does not read the hdf5 Nexus Geometry files.

KarlPalmen, 2018-03-14:
OK I'll send him an E-mail to check this. This example file provided by @matthew-d-jones is Nexus file, that is like an ISIS nexus file except it has an instrument with just a name and a sample containing the shape in 3 data blocks that apparently match the 3 arguments of `nexusShapeFactory.createFromOFFMesh`.

Anders-Markvardsen, 2018-03-21:
Hi @KarlPalmen. Based on the discussion above I would limit the scope of this issue to extend loadshape to read a single OFF file and the OFF format as described on wikipedia. 

As a side, from Karl's comment above, @matthew-d-jones I find it worrying that the OFF format used in nexus files differs from that of the OFF format on wikipedia. Is there a good reason for this?

matthew-d-jones, 2018-03-21:
@Anders-Markvardsen @KarlPalmen 
The issue title mentions the NeXus OFF-based format and the description instead mentions the OFF format, this is confusing.

The NeXus `NXoff_geometry` is based on the OFF format and is very close to it. The difference is an extra dataset which indexes into a flattened dataset containing the vertices for each of the faces. This avoids the need for a ragged-edged faces dataset, which is desirable as ragged-edged datasets are difficult to work with and best avoided in HDF5 files.
This will be described in the NeXus manual (there is a PR waiting), here is a pdf of the design page, section 1.2.2.3 is the relevant one.
[nexus_manual.pdf](https://github.com/mantidproject/mantid/files/1833088/nexus_manual.pdf)

KarlPalmen, 2018-03-21:
It was my guess that faces of varying number of vertices could be supported, but ragged datasets were to be avoided.

KarlPalmen, 2018-03-22:
It would be much quicker for me to load an ASCII OFF file. I have not yet figured out how to use the Nexus/HDF objects to load a Nexus file. Code in other Nexus loading algorithms is too complicated to be helpful.

KarlPalmen, 2018-03-22:
Pull request #22016 needs merging, before coding work on this issue can proceed.

PR #22385 (2018-05-21): LoadSampleShape that reads Ascii OFF files
**Description of work**
Extend LoadSampleShape to work with ASCII OFF files as described at 
https://en.wikipedia.org/wiki/OFF_(file_format) 

**To test:**
Run MantidPlot to obtain a workspace with an instrument and then run LoadSampleShape on one of the
[off.zip](https://github.com/mantidproject/mantid/files/1974044/off.zip) files attached or a modification of such a file made by you.

I found some more OFF files at
http://people.sc.fsu.edu/~jburkardt/data/off/off.html
Of these `cone.off, icosa.off, mctet.off, mctri.off, octa.off, socket.off & tetra. [... cut]
closed 2018-05-21 by martyngigg