mantidproject/mantid #21573: Document differences between muon-specific maxent alg and vanilla maxent alg

Anders-Markvardsen, 2018-01-17:
A muon-specific maxent+extra algorithm has been translated from fortran to python (by instrument scientist James Lord) and is available from the script repository: 

`ScriptRepository/Muon/OpengenieMaxent.py is the main algorithm.`
`ScriptRepository/Muon/OpengenieMaxentTools/*.py are the various Fortran subroutines each still in its own file for ease of debugging.`

`Make sure .../ScriptRepository/muon is included in the Python Extensions/Algorithms list. There's no need to include .../ScriptRepository/muon/ OpengenieMaxentTools.`

It has a number of features which are different from the vanilla Mantid MaxEnt algorithm (http://docs.mantidproject.org/nightly/algorithms/MaxEnt-v1.html):

* It modify the input data in an inner maxent loop in a muon specific way, using response function R(f), deadtime correction, exponential decay and a phase value for each detector (possibly more). Whereas, the vanilla Mantid MaxEnt alg assumed that the input data has been fully corrected
* It has an outer loop which allow muon specific parameters used in the inner loop to be fitting
* It assumes the frequency spectrum, which the algorithm tries to reconstruct is real and positive. For the vanilla MaxEnt this correspond to setting PositiveImage=True.
* It is setup only to reconstruction from many spectra and with the right input such as one phase value for each detector
* Has little documentation, and the documentation available assumed good understanding of muon science
* Including it contains little documentation of its generic maxent specific parameters are. In OpengenieMaxent.py these are 'DefaultLevel' and 'Factor'. In vanilla Mantid maxent these are probably closely related to A and ChiTargetOverN

In this issue:

* Write a python script that uses both of these maxent algorithms, but takes exactly the same data input. The vanilla Maxent works on spectrum at a time, therefore the data input must be one spectrum. The data may be created as simulated time-domain data transformed from frequency-domain model data. When using OpengenieMaxent all usage of deadtime, remove exponential decay etc must be bypass to be able to compare the two maxent algorithms. Report final version of this script in this issue
* Adjust the above described OpengenieMaxent.py parameters 'DefaultLevel' and 'Factor' and MaxEnt parameter 'A' and 'ChiTargetOverN' till you get good agreement between these two algorithms and report final finding of this in this issue, if many images - collect as in a word document first
* Look into OpengenieMaxent.py and compare with the algorithm documentation of MaxEnt and explain how 'DefaultLevel' and 'Factor' relates to  'A' and 'ChiTargetOverN'. Report final findings of this in this issue

KarlPalmen, 2018-03-22:
I don't understand the sentence
 `•Including it contains little documentation of its generic maxent specific parameters are` 
It has syntax errors I can't resolve.

KarlPalmen, 2018-03-22:
I've fixed the grammar in the description, except for the sentence, I've just mentioned, please verify, it says the correct things.

KarlPalmen, 2018-03-22:
A quick look at the script repository shows that `OpenGenie` has been removed from the names and the main script has been divided into model, presenter and view (`maxent_model.py` etc.).

KarlPalmen, 2018-04-05:
How does run the script?

Is it possible to modify the script for issue #21444 to run this algorithm instead?

Anders-Markvardsen, 2018-04-05:
Since this issue was created, the muon-specific maxent code discussed in the description of this issue has now been wrapped up into the Mantid algorithm: http://docs.mantidproject.org/nightly/algorithms/MuonMaxent-v1.html. 

Hence, everywhere you says OpengenieMaxent above you should read as MuonMaxent. 

The task of this issue is still the same, but it is to compare the two algorithms MuonMaxent and MaxEnt.

KarlPalmen, 2018-04-05:
Thank you for letting me know. I found the algorithm document in a recent build of the documentation.

Anders-Markvardsen, 2018-05-17:
See below for questions and comments to your MMComparison.docx

[MMComparison_with_comments.docx](https://github.com/mantidproject/mantid/files/2012598/MMComparison_with_comments.docx)

Anders-Markvardsen, 2018-05-18:
Here is a Python script answering some of questions in the word document above, and incrementally move towards creating an example which compares apples with apples:

```
#
#
# Example 1: where take as input into vanilla MaxEnt data with exponential Muon decay. This
# may cause MaxEnt to be slower as it will not to reconstruct frequencies values to describe this exponential decay also
# MuonMaxEnt multiply internally an exponential decay just before comparing with the data in its chi-squared expression. This 
# operation means, in effect, the reconstruction is against data with no exponential decay 
#
#

# load a raw muon data set which contains an exponential decay
ws1 = Load(r'C:\Users\ajm64\Dropbox\ISIS Projects\maxent_karl\MuonMaxentTest.nxs')

# The data are cropped to starting value Xmin=0.1 to replicate MuonMaxent default for FirstGoodTime=0.1 
ws1c = CropWorkspace(InputWorkspace=ws1,Xmin=0.1 ,Xmax=10)

# Run vanilla MaxEnt on raw cropped workspace
evolChiP, evolAngleP, imageP, dataP = MaxEnt(InputWorkspace='ws1c', ComplexData=False, A=0.2, MaxIterations=5000, ChiTargetOverN = 0.50, PositiveImage=True)

# Run MuonMaxent
#
# This comparison is flawed for at least the following reasons:
#     1. MaxEnt also reconstruct the exponential decay but with MuonMaxent you don't
#     2. How does Npts=16384 + MaxField=1000 compare with equivalent set in MaxEnt
#     3. Why is OuterIterations='9' and not OuterIterations='1'
#     4. No initial phase is input to MuonMaxent, i.e. no InputPhaseTable values, what does that mean?
#     5. Should MuonMaxent FixPhases not be True for this comparison?
#
#
MuonMaxent(InputWorkspace='ws1 ', LastGoodTime=10, Npts=16384 , MaxField=1000 , FitDeadtime=False,OuterIterations='9' , InnerIterations='12', DefaultLevel=0.5, Factor=0.2 , OutputWorkspace='imagePmuon', OutputPhaseTable='phasesOut', ReconstructedSpectra='dataPmuon')

# Spectra to compare for this example:
#  - dataP and data dataPmuon

# For the above reasons this first comparison of the maximum entropy algorithms does not compare apples with apples

#
# 
# Example 2: incremental improvement to Example 1. Addressing issue 1 and 3 above
#  1. Remove exponential decay before running MaxEnt. This ensures that the two maximum entropy algorithms are both 
#     not reconstructing an exponential decay
#  3. put OuterIterations='1'
#
#

# remove exponential decay from cropped workspace above
ws2R = RemoveExpDecay('ws1c')

# run MaxEnt where
evolChiP2, evolAngleP2, imageP2, dataP2 = MaxEnt(InputWorkspace='ws2R', ComplexData=False, A=0.4, MaxIterations=20000, ChiTargetOverN = 1.0, PositiveImage=True)

# rescale frequency axis from inverse time to magnetic field to better compare with output from MuonMaxent
imageP_field2 = ScaleX("imageP2",Factor=1.e3/13.55)

# run MuonMaxent of raw data. Before MuonMaxent compared reconstructed data with the real data it multiplies the
# reconstructed data with a muon exponential
MuonMaxent(InputWorkspace='ws1', LastGoodTime=10, Npts=16384, MaxField=1000, FitDeadtime=False,OuterIterations='1', InnerIterations='12', DefaultLevel=0.5, Factor=0.2, OutputWorkspace='imagePmuon2', OutputPhaseTable='phasesOut2', ReconstructedSpectra='dataPmuon2')

# Remove exponential decay from the MuonMaxent reconstructed time spectrum to be able to compared it with the
# MaxEnt reconstructed time spectrum
dataPmuon2_removeExpDecay = RemoveExpDecay('dataPmuon2')

# Spectra to compare for this example:
#   - time domain: dataPmuon2_removeExpDecay and real part of dataP2
#   - frequency domain: imageP_field2 and imageP2
```

Anders-Markvardsen, 2018-05-18:
Above script also here attached as a zip file, and please replace the absolute path for loading the data

[comparing.zip](https://github.com/mantidproject/mantid/files/2016406/comparing.zip)

KarlPalmen, 2018-05-18:
Dear Anders

I have just finished replying to your comments on the document and attach here. I’ll next try out the script you have just sent.

Karl

KarlPalmen, 2018-05-18:
Here add my comments here 
[MMComparison_with_comments.docx](https://github.com/mantidproject/mantid/files/2017408/MMComparison_with_comments.docx)

KarlPalmen, 2018-05-18:
Points 2, 4 & 5 below need addressing
```
 This comparison is flawed for at least the following reasons:
     1. MaxEnt also reconstruct the exponential decay but with MuonMaxent you don't
     2. How does Npts=16384 + MaxField=1000 compare with equivalent set in MaxEnt
     3. Why is OuterIterations='9' and not OuterIterations='1'
     4. No initial phase is input to MuonMaxent, i.e. no InputPhaseTable values, what does that mean?
     5. Should MuonMaxent FixPhases not be True for this comparison?
```
For 2, I don't know what the equivalent set in MaxEnt is.
For 5, if set TRUE, then a phase table is needed.
For 4, I don't know. This is related to 5.

closed 2021-03-02 by stale[bot]