mantidproject/mantid #21577: MaxEnt: allow phases for datasets

Anders-Markvardsen, 2018-01-18:
Following completion of #21573 add the equivalent a new algorithm property to MaxEnt which allows to do the same as that of inputting phase detector values into OpengenieMaxent.py, but done in a way that it is not muon-specific. 

Further:

* Add documentation + usage example demonstrating this
* Demonstrate that the results with phases in MaxEnt matches that of OpengenieMaxent.py

When this option is used, on say N input datasets, the information of these data are combined and only one dataset is returned.

KarlPalmen, 2018-06-14:
Dear @Anders-Markvardsen & @AnthonyLim23   Should `PerSpectrumReconstruction` be `perSpectrumImage`, because if false, it would still per spectrum reconstruction, but using one image and with reconstructions differening only from each other by application of the adjustments?

AnthonyLim23, 2018-06-14:
yes, I think `perSpectrumImage` is a better name

KarlPalmen, 2018-06-15:
I've thought more carefully about this and realise I still do not fully understand what is required wher perSpectrumImage is set false. In particular I don't understand what is meant by "one image is used to fit against all spectra simultaniously". Does this mean a single fitting and that reconstructed images are then created by apply the adjustments for each spectrum afterwards? Or does it mean a separate fitting is done with each spectrum with the image somehow made the same. I never grasped the MuonMaxEnt algorithm.

KarlPalmen, 2018-06-15:
It looks like the Chi squared is calculated as though to spectra were concatenated, but the image is calculated with the sum of a spectra. This cannot be handled by the existing `MaxentCalculator` class.

KarlPalmen, 2018-06-15:
A talk with Anders confirms that the Chi square is calculated as though the spectra were concatenated into one spectrum.

KarlPalmen, 2018-06-18:
I'll go ahead of plan to create new class to run in place of `MaxentTransformFourier' and inheriting from it. The class will take data as concatenated spectra, while having a single image. The spectra in data will be added together when transforming to image. When transforming to data a concatenated copies of the reconstructed data will be created (one fpr each spectrum) and the adjustements will be applied to them.

The spectra in the data are concatenated so they can be handled by the existing Chi squared calculations in `MaxentCalculator`.

PR #22558 (2018-09-26): Add adjustments to reconstructed data in Maxent
**Description of work.**

**To test:**

Run the algorthm with the new properties DataLinearAdj, DataConstAdj & PerSpectrumReconstruction and check it behaves as expected. Also build and check the documentation of Maxent.

I suggest running variations of the two usage examples added to the documentation to the documentation to do the testing.

Fixes #21577 

Does this update require release notes?
- [x] Yes
- [ ] No
 

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testin [... cut]
closed 2018-09-26 by soininen