mantidproject/mantid #21605: Add ILL training materials

ianbush, 2018-01-22:
This PR will add a section in the sphinx documentation for ILL training sessions. There should be four separate issues following this one, with branches based off the ones for this issue to cover the planned training, similar to the courses below.

[Mantid Basic Course](http://www.mantidproject.org/Mantid_Basic_Course)
[Python In Mantid](http://www.mantidproject.org/Python_In_Mantid)

Approximate plan:

**1. Introduction**
* Structure for training session
* Downloading and installing (release, nightly), setting up data directories, usage data, supported platforms
* Sources of help - CS group, online forums, training materials on website, algorithm documentation, API documentation
* Bug reporting - please do it!
* Structure of Mantid - Python API written in C++/Python, MantidPlot as current GUI, new workbench coming eventually

**2. Basics of MantidPlot**
* Loading data from NeXus files, loading multiple files (important for ILL)
* Viewing the workspace data, monitors
* Plotting a spectrum/spectra
* 2D plots
* SpectrumViewer
* Using the instrument view

**3. More on Mantid(Plot)**
* Workspaces - basic types, historgram/point data
* Detector table - spectrum numbers, workspace indexes, detector IDs
* Algorithms - how to use (input validation, workspace/file selection), dialogs, workspace history
* Saving out workspaces to NeXus, other formats
* Interfaces - where to find, fitting as example?

**4. Python and Mantid**
* How to use - from mantid.simpleapi import *`, script window
* Running an algorithm
* Generating a script from history
* Getting workspaces in MantidPlot `mtd['name]`
* Very basic workspaces data access via NumPy arrays
* Possibility to write own algorithms (mention only)

gvardany, 2018-01-25:
@soininen @verena I have just added a bullet to your parts concerning workspace groups, I think this is also important to teach.

reimundILL, 2018-01-25:
`Generating a script from history` can be covered when talking about workspace histories. It is just the clipboard content containing Python code.

PR #21714 (2018-02-07): Sphinx training docs
Add some training materials to the Sphinx documentation.

Currently these are written in an ILL specific way. The long term aim will be to make these generic training documents, where different facilities can easily pick and choose which parts to use. Time limitations has meant this is as far as we have got for now.

**To test:**

Compile and read the documentation.

Fixes #21605.

**Release Notes** 
*Does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testin [... cut]
closed 2018-02-07 by rosswhitfield