mantidproject/mantid #21611: Investigate AlgorithmSelector sizing issue

ElliotAOram, 2018-01-22:
The `AlgorithmSelector` widget in the workbench has a bug with the minimum size. The sizing policy doesn't seem to be properly initialised upon creation of the layout. At the moment we are calling `layout.sizeHint()` to get round this. This issue should cover investigating and implementing a permanent solution

suspect code can be found in `init_ui(self):` within file `\mantid\qt\python\mantidqt\widgets\algorithmselector\widget.py`

DTasev, 2019-01-15:
I tried looking into this but I can't replicate it. I don't see a difference in size or resizing in the AlgorithmSelector widget with or without the `sizeHint` call. Do you remember more details about the issue? I read the referenced PR and it seems like it was something to do with AlgorithmSelector widget not being resizable

ElliotAOram, 2019-01-28:
Willing to gamble that this is out of date now. I can't remember the specifics as it was over a year ago but I think it was a fairly obvious visible bug. If you can't reproduce this I think it's fine to close it.

closed 2019-01-28 by DTasev