mantidproject/mantid #21619: FitFunction TeixeiraWater lacks Planck constant

MargheF, 2018-01-22:
FitFunction "TeixeiraWater" models the Q dependence of the QENS line width (Gamma (hwhm)) as
hwhm = xvals * xvals * length / (tau * (1 + xvals * xvals * length))
where xvals is Q and length is L^2

Shouldn't be like that:
hwhm = hbar * xvals * xvals * length / (tau * (6 + xvals * xvals * length)) ?

HWHM contains hbar as it's presented in FitFunction "TeixeiraWaterSQE".
Also, if diffusion coefficient D=L^2/(6*tau), then 6 goes in place of 1.

jmborr, 2018-01-22:
@samueljackson92 I think you wrote this function, right? Could you please address this issue?

samueljackson92, 2018-01-23:
I think I added it, but that it was originally written by @SpencerHowells, who may want to comment? I agree that it does look like the diffusion constant is incorrect here.

MargheF, 2018-01-23:
A lack of Planck constant is possible to overcome by dividing Gamma into hbar. However, as diffusion coefficient D=L^2/(6*tau), then 6 goes in place of 1 in original:
hwhm = xvals * xvals * length / (tau * (1 + xvals * xvals * length))
Could you, please, modify this? I am not allowed to change the script.
Also, is the fitting parameter "L" a length (i.e. square root of the "length" in the formula)?

SpencerHowells, 2018-01-24:
@MargheF @I have contacted our QENS scientists to confirm this is what they want. Then we can update.

MargheF, 2018-02-01:
@samueljackson92 Thank you for fixing definition of fit function. **Is the change already effective?** Sorry, I am not very good with programming; could you, please, give me a hint on this line:
expected = [3.74985938e-05,   3.73599004e-03,   2.72727273e-01,   9.74025974e-01]
**What's "expected"?** Was hbar introduced to the fit function? Also, fitting parameter "L" is square root of length**2 or is it actually length**2?
Thank you very much!

PR #21744 (2018-04-25): FitFunction TeixeiraWater lacks Planck constant
Description of work.

**To test:**

Fixes #21619

**Release Notes** 

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the code conform to the [coding standards](http://www.mantidproject.org/Coding_Standards)?
- [ ] Are the unit tests small and test the class in isolation?
- [ ] If there are changes in the release notes then do they describe the changes appropriately?

##### Functio [... cut]
closed 2018-04-25 by martyngigg