mantidproject/mantid #21620: Memory leak when using setX

rosswhitfield, 2018-01-22:
@martyngigg I'm guessing this was introduced with #21407

When I run the following ~6Gb of memory gets used every iteration. The `3.11.0` release doesn't have this problem.
```
from mantid.simpleapi import *
import numpy as np
import resource

w = np.array([1.487,1.489])

for n in range(10):
    print(n,resource.getrusage(resource.RUSAGE_SELF).ru_maxrss)
    ws=LoadEmptyInstrument(InstrumentName='WAND')
    for idx in xrange(ws.getNumberHistograms()):
        ws.setX(idx, w)
    DeleteWorkspace(ws)
```

Stable release shows no memory increase
```
$ mantidpython leak.py
FrameworkManager-[Notice] Welcome to Mantid 3.11.0
FrameworkManager-[Notice] Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid3.11
DownloadInstrument-[Notice] All instrument definitions up to date
(0, 157804)
LoadEmptyInstrument-[Notice] LoadEmptyInstrument started
LoadEmptyInstrument-[Notice] LoadEmptyInstrument successful, Duration 17.18 seconds
DeleteWorkspace-[Notice] DeleteWorkspace started
DeleteWorkspace-[Notice] DeleteWorkspace successful, Duration 0.00 seconds
(1, 2240980)
LoadEmptyInstrument-[Notice] LoadEmptyInstrument started
LoadEmptyInstrument-[Notice] LoadEmptyInstrument successful, Duration 2.51 seconds
DeleteWorkspace-[Notice] DeleteWorkspace started
DeleteWorkspace-[Notice] DeleteWorkspace successful, Duration 0.00 seconds
(2, 2241720)
LoadEmptyInstrument-[Notice] LoadEmptyInstrument started
LoadEmptyInstrument-[Notice] LoadEmptyInstrument successful, Duration 2.40 seconds
DeleteWorkspace-[Notice] DeleteWorkspace started
DeleteWorkspace-[Notice] DeleteWorkspace successful, Duration 0.00 seconds
(3, 2241720)
LoadEmptyInstrument-[Notice] LoadEmptyInstrument started
LoadEmptyInstrument-[Notice] LoadEmptyInstrument successful, Duration 2.39 seconds
DeleteWorkspace-[Notice] DeleteWorkspace started
DeleteWorkspace-[Notice] DeleteWorkspace successful, Duration 0.00 seconds
(4, 2241720)
...
```

But with nightly the memory use just keeps increasing
```
$ mantidpythonnightly leak.py
FrameworkManager-[Notice] Welcome to Mantid 3.11.20180104.1953
FrameworkManager-[Notice] Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid
(0, 152768)
LoadEmptyInstrument-[Notice] LoadEmptyInstrument started
DownloadInstrument-[Notice] All instrument definitions up to date
LoadEmptyInstrument-[Notice] LoadEmptyInstrument successful, Duration 16.49 seconds
DeleteWorkspace-[Notice] DeleteWorkspace started
DeleteWorkspace-[Notice] DeleteWorkspace successful, Duration 0.00 seconds
(1, 8172776)
LoadEmptyInstrument-[Notice] LoadEmptyInstrument started
LoadEmptyInstrument-[Notice] LoadEmptyInstrument successful, Duration 1.81 seconds
DeleteWorkspace-[Notice] DeleteWorkspace started
DeleteWorkspace-[Notice] DeleteWorkspace successful, Duration 0.00 seconds
(2, 14089876)
LoadEmptyInstrument-[Notice] LoadEmptyInstrument started
LoadEmptyInstrument-[Notice] LoadEmptyInstrument successful, Duration 1.83 seconds
DeleteWorkspace-[Notice] DeleteWorkspace started
DeleteWorkspace-[Notice] DeleteWorkspace successful, Duration 0.00 seconds
(3, 20003964)
LoadEmptyInstrument-[Notice] LoadEmptyInstrument started
LoadEmptyInstrument-[Notice] LoadEmptyInstrument successful, Duration 1.84 seconds
DeleteWorkspace-[Notice] DeleteWorkspace started
DeleteWorkspace-[Notice] DeleteWorkspace successful, Duration 0.00 seconds
(4, 25924340)
LoadEmptyInstrument-[Notice] LoadEmptyInstrument started
LoadEmptyInstrument-[Notice] LoadEmptyInstrument successful, Duration 1.83 seconds
DeleteWorkspace-[Notice] DeleteWorkspace started
DeleteWorkspace-[Notice] DeleteWorkspace successful, Duration 0.00 seconds
(5, 31839376)
LoadEmptyInstrument-[Notice] LoadEmptyInstrument started
LoadEmptyInstrument-[Notice] LoadEmptyInstrument successful, Duration 1.81 seconds
DeleteWorkspace-[Notice] DeleteWorkspace started
DeleteWorkspace-[Notice] DeleteWorkspace successful, Duration 0.00 seconds
(6, 37755768)
LoadEmptyInstrument-[Notice] LoadEmptyInstrument started
LoadEmptyInstrument-[Notice] LoadEmptyInstrument successful, Duration 1.77 seconds
DeleteWorkspace-[Notice] DeleteWorkspace started
DeleteWorkspace-[Notice] DeleteWorkspace successful, Duration 0.00 seconds
(7, 43680192)
LoadEmptyInstrument-[Notice] LoadEmptyInstrument started
LoadEmptyInstrument-[Notice] LoadEmptyInstrument successful, Duration 1.79 seconds
DeleteWorkspace-[Notice] DeleteWorkspace started
DeleteWorkspace-[Notice] DeleteWorkspace successful, Duration 0.00 seconds
(8, 49600536)
LoadEmptyInstrument-[Notice] LoadEmptyInstrument started
LoadEmptyInstrument-[Notice] LoadEmptyInstrument successful, Duration 1.75 seconds
DeleteWorkspace-[Notice] DeleteWorkspace started
DeleteWorkspace-[Notice] DeleteWorkspace successful, Duration 0.00 seconds
(9, 55509908)
LoadEmptyInstrument-[Notice] LoadEmptyInstrument started
LoadEmptyInstrument-[Notice] LoadEmptyInstrument successful, Duration 1.76 seconds
DeleteWorkspace-[Notice] DeleteWorkspace started
DeleteWorkspace-[Notice] DeleteWorkspace successful, Duration 0.00 seconds
```

martyngigg, 2018-01-23:
@rosswhitfield I think you're right. That issue is the most likely cause. Can this wait until next week for me to look at it?

rosswhitfield, 2018-01-23:
@martyngigg Yep, no problem. It just should probably be fixed before the release.

martyngigg, 2018-01-23:
Definitely. Once we get the workshop demos out the way I can think about it then.

PR #21661 (2018-01-29): Fix memory leak in Python setX/Y/E workspace methods
Description of work.

Uses the `boost::python::object` wrappers to correctly handle the reference count of the new array iterator used to copy from the ndarray to the container.

The reference count was definitely being mishandled but rather confusingly the line that was the problem was also incorrect in the last release (and for years as far as I can see) yet the problem did not appear. I assume this is some `numpy` internal black magic that I do not understand. 

**To test:**

Use the script from the issue to confirm that the memory usage  [... cut]
closed 2018-01-29 by SimonHeybrock