mantidproject/mantid #21626: Incomplete Algorithm time reporting.

SimonHeybrock, 2018-01-23:
`Algorithm` does not include property validation and other setup code in the reported time. This can lead to large confusing discrepancies between observed and logged time.

OwenArnold, 2018-01-24:
@SimonHeybrock I think the problem you describe is related to this #20445.

SimonHeybrock, 2018-01-24:
@OwenArnold I tried the example and the reported time still looks not complete. Probably due to things happening in `simpleapi` before the C++ `Algorithm::execute` is called. So I think I cannot claim to have fixed that issue (only partially).

SimonHeybrock, 2018-01-24:
@OwenArnold Now this is weird. If I set the log level to `information`, the reported timings of `CreateWorkspace` look actually correct. However, setting it to `debug` Mantid freezes for about 2 minutes to execute the script, but does not report the time. Note that there is basically no debug output, so I have no idea why it suddenly takes so long?

SimonHeybrock, 2018-01-24:
Correction: There are basically no debug output *lines*, but there are a couple very long ones containing the unwrapped `range(1000000)`. Apparently this is causing the issue.

Conclusion: I am fixing #20445, the long runtime with debug output is unrelated (and probably parts of the problem reported in #20443).

OwenArnold, 2018-01-24:
Yes, looking at other algorithms. The sum of timings match the reported total. I do not see this for `CreateMDWorkspace`

PR #21627 (2018-01-24): Include time spent in Algorithm::execute in reported time.
See issue description. Also add debug output regarding time within execute.

**To test:**

Code review.

Fixes #21626.
Fixes #20445.

**Release Notes** 

Updated.

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the code conform to the [coding standards](http://www.mantidproject.org/Coding_Standards)?
- [ ] Are the unit tests small and test the class in isolation?
- [ ]  [... cut]
closed 2018-01-24 by peterfpeterson