mantidproject/mantid #21649: Umbrella Issue for Workbench Bugs

martyngigg, 2018-01-25:
This issue contains a list of bugs discovered during the demos with existing features in the new workbench.

AnthonyLim23, 2018-01-25:
when using an additional monitor (to a laptop) the resolution gets messed up 
[image]

rosswhitfield, 2018-01-25:
`codeeditor` doesn't work with `IPython>=6.0`

When I try to execute anything I get
```
TypeError: _update_indent() takes 1 positional argument but 2 were given
  File "/home/rwp/mantid/qt/python/mantidqt/widgets/codeeditor/execution.py", line 222, in code_blocks
    isp.push(line)
  File "/home/rwp/mantid/qt/python/mantidqt/widgets/codeeditor/inputsplitter.py", line 71, in push
    self._update_indent(lines)
```

They changed `._update_indent()` to not take arguments, see [here](https://github.com/ipython/ipython/blob/6.x/IPython/core/inputsplitter.py#L494)

But also in `6.3` they then removed `._update_indent()` and replaced it with `.get_indent_spaces()`. See ipython/ipython#10898

martyngigg, 2018-01-26:
@rosswhitfield Thanks. The price paid for using "private" methods.

AnthonyLim23, 2018-01-26:
in the figure options changing the face colour of the marker to` 'g' `will crash the workbench

martyngigg, 2018-01-26:
Thanks, Anthony.

martyngigg, 2018-01-26:
An issue with IPython >= 5 and the startup banner. The whole thing prints twice...
```
IPython 5.3.0 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.

Python 3.6.3 (default, Oct  9 2017, 12:11:29), numpy 1.12.1, matplotlib 2.0.0
Type "copyright", "credits" or "license" for more information.

The following code has been executed at startup:

from __future__ import (absolute_import, division, print_function, unicode_literals)
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np
Python 3.6.3 (default, Oct  9 2017, 12:11:29) 
Type "copyright", "credits" or "license" for more information.

IPython 5.3.0 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.
```

rosswhitfield, 2018-01-26:
`IPython.core.usage.quick_guide` was removed in IPython `6.0`. See ipython/ipython#9912

ianbush, 2018-01-29:
There are locale issues when your system has different number settings. To reproduce on Ubuntu 16.04 go to System Settings -> Language Support -> Regional Formats. Choose a setting for numbers which uses a comma for the decimal separator, e.g francais. Then make a plot in the workbench and go to settings. You can no longer enter decimal points, but using commas causes a hard crash. The OK button seems to be disabled as soon as something is edited and does not come back,

When the locale _is_ a point for the decimal separator commas can still be entered which causes a crash.

ianbush, 2018-01-29:
When you run something in the Python script editor by selecting a small part of the full script the line numbers in the message output and the red arrow goes to the wrong place.

E.g. open a new script window and change the import matplotlib (line 8) to something that does not exist, e.g. `import blah as bl`. Highlight lines 7 - 10, then run the script. Check the message window and see the arrow points to the wrong place.

ianbush, 2018-01-30:
Set default background colour to white for matplotlib - it looks much nicer (default in 2.1.2).

martyngigg, 2019-04-16:
Other issues have now been created to cover this or these issues have been solved. See the project board.

closed 2019-04-16 by martyngigg