mantidproject/mantid #21735: New MAPS instrument definition

fodblog, 2018-02-12:
Not a bug report. A request from a user to update an instrument definition file. The content of the request is below:

Back in December we modified the MAPS IDF following an upgrade to the instrument. After some further careful checking we’ve had to change a few of the parameters therein.

abuts, 2018-02-13:
@keeeto Dear Keith, 
MAPS users are waiting for the results of this ticket and as I understand, you already have proper MAPS_definition file. Could you, please, start branch and commit the file for me to be able to modify MAPS_parameters for reduction to work properly.

keeeto, 2018-02-13:
@abuts there is a branch with the updated file `21735_new_maps_idf`(you can always look for the branch by typing the issue number into branches on the repo). If you can add the MAPS_parameters that would more or less finish this up. Then I'd just need to run some tests (I am waiting for data to do this) and I can submit a pull request. Do you happen to have any `.nxs` files that I could use to run some tests?

abuts, 2018-02-13:
@keeeto We can pick up the recent data from MAPS but I would not bother as Russell did the testing and changes so he is responsible. We just need to ensure it not contradicts anything in Mantid. I will take the branch and verify the reduction settings. 
Regards,
Alex

keeeto, 2018-02-13:
That's great thanks Alex. Can you send me some MAPS data files anyway - I am trying to build up a library of test files for as many instruments as possible, so it would be extremely helpful. Meantime, I'll wait for your commit to the branch then I can submit a pull request and one of the gatekeepers can decide when to merge into the main code.

abuts, 2018-02-13:
@keeeto  this would not work and the tests will fail. The changes will break the reduction and all data for all previous runs. I am preparing proper changes.

abuts, 2018-02-13:
@keeeto  -- I am looking at the changes more carefully and do some comparison. 
Are you in ISIS or elsewhere (for better way of distributing the  data)

keeeto, 2018-02-13:
@abuts -- I am in ISIS. I can come and meet you, if that works.

keeeto, 2018-02-13:
@abuts no problem. We could perform the tests that we wanted on different data. Basically we wanted to test that Mantid uses the updated IDF if there is an old IDF in the install directory and a new IDF in the download directory and both have the same dates in them. We can confirm that Mantid behaves as desired and uses the IDF from the download directory. If you update the parameters file on the branch and push the commit, just let me know. I will then make a pull request.

keeeto, 2018-02-13:
@abuts -- Thanks for the information. 

Do you know what the name of the unit test that checks for the overlapping dates is? It would be good to make sure that it is not picking up 'errors' that it should not pick up.

abuts, 2018-02-13:
@keeeto, no, unfortunately I don't remember, just remember having some test failing when I was erroneously provided an overlapping dates. Martin does certainly know better.  It may be doc test or system test.

keeeto, 2018-02-13:
@abuts -- no problem, we can work around this. I see that you have pushed some changes up. Are you finished editing? If so, I will go ahead and create a pull request.

abuts, 2018-02-13:
@keeeto yes, I think this should be it

keeeto, 2018-02-13:
@abuts -- are you finished updating this branch? I had to make some modifications to the git history for this branch as there were old commits that I needed to remove. When you merge in your changes the old commits came back and the automatic tests will fail. So I just need to know that you are fully finished before starting the pull request again.

keeeto, 2018-02-13:
Yes, that flake warning. It relates to a GUI file that I was working on, which should not be in this branch, but I committed by accident. I can remove that commit easily enough, but merging in brings it back.

abuts, 2018-02-13:
@keeeto I should fix flake and if it goes well you can do whatever you want with it. I do not have local changes so you can force push and I will just delete my branch if necessary.

keeeto, 2018-02-13:
@abuts -- Okay that's sorted out now. Hopefully the pull request will go through smoothly enough!

closed 2018-02-27 by keeeto