mantidproject/mantid #21755: Filtering of monitor events broken in LoadEventNexus

martyngigg, 2018-02-15:
It appears that monitor events are no longer filtered by time if that is requested. I think this was broken in #15667. The old behaviour used the same time filtering for the main data and monitors. Now that `LoadNexusMonitors` is doing the loading it ignores the filter times.

In addition the 4 properties `FilterMonByTofMin`/`FilterMonByTofMax`/`FilterMonByTimeStart`/`FilterMonByTimeEnd` do not seem to do anything at all in the code.

Finally,  ticking the option `MonitorAsEvents` & `LoadEventMonitors` results in zero events in the monitor workspace - these options are also quite confusing in how they are supposed to related to each other. Should `MonitorAsEvents` be deprecated?

### Expected behavior

```python
LoadEventNexus(Filename='/home/dmn58364/Code/mantidproject/mantid/builds/debug/ExternalData/Testing/Data/SystemTest/LET00006278.nxs', OutputWorkspace='LET00006278', LoadMonitors=True, LoadEventMonitors=True)
FilterByTime(InputWorkspace='LET00006278_monitors', OutputWorkspace='LET00006278_monitors',
StartTime=0, StopTime=60)
```

should be equivalent to:

```python
LoadEventNexus(Filename='/home/dmn58364/Code/mantidproject/mantid/builds/debug/ExternalData/Testing/Data/SystemTest/LET00006278.nxs', OutputWorkspace='LET00006278', LoadMonitors=True, LoadEventMonitors=True, FilterByTimeStart=0, FilterByTimeStop=60)
```

### Actual behavior

The number of events in the monitor workspace remains constant regardless of the filtering options. I think by default the monitors should be filtered by the same time as main data *unless* one of the `FilterMon*` options is provided.

martyngigg, 2018-02-15:
cc @peterfpeterson

NickDraper, 2018-02-28:
@wdzhou If this is actively being worked on and is likely to be available in the next few days then move this back to 3.12

NickDraper, 2018-07-05:
Neither  high priority under the new guidelines, or in progress under anyone's definition

NickDraper, 2018-07-06:
This has been superceeded by the work done by @peterfpeterson

peterfpeterson, 2018-07-06:
This is still needed. #22606 did not add this functionality back into `LoadNexusMonitors`.

closed 2021-03-02 by stale[bot]