mantidproject/mantid #21786: Documentation updates from dev meeting

keeeto, 2018-02-20:
## Suggested updates to documentation

### Re-order algorithm documentation
* Re-order by category, not just index

### Date stamp at bottom of algorithm page
* Source date
* Docs date

### Documentation link
* The link at the top of each page should bring you to http://docs.mantidproject.org/nightly/
* Add sections to  http://docs.mantidproject.org/nightly/ for:
    * Beginners guides
    * Examples
    * Concepts

DanNixon, 2018-02-23:
I have the source Git timestamp working, let me know what format you want it displayed in and I'll have a PR ready.

keeeto, 2018-02-23:
Oh, that's much better than the one I put it, which I think is just when Sphinx last built. I already made a PR on this. Do you want to update the existing PR, or put in a new one? 

I went with standard %b %d, %Y (Month day, Year) format.

DanNixon, 2018-02-23:
I can do it as a separate one if you like.

Note mine does the date of the last commit that edited algorithm, fit function, etc sources for displaying in the source links at the bottom of the pages.

I did try and do a similar one for the documentation pages themselves that could be applied to all documentation pages but didn't get too far with it.

keeeto, 2018-02-23:
That sounds great. I wouldn't worry about the documentation too much. Note Karl's contention with the date formatting on the PR. We will go with '%Y-%m-%d' (YYYY-mm-dd) format instead.

DanNixon, 2018-02-23:
Cool, I'll get it done next week now.

closed 2018-02-27 by keeeto