mantidproject/mantid #21836: Some sample log has zero entry after FilterEvents

wdzhou, 2018-02-22:
Here is the script

```
LoadEventNexus(Filename='/SNS/REF_M/shared/ADARA.Test.Data.2018/REF_M_28131.nxs.h5',
               OutputWorkspace='test31')
               
GenerateEventsFilter(InputWorkspace='test31', OutputWorkspace='filter', InformationWorkspace='filter_info', LogName='SF1',
                     MinimumLogValue=0.9, MaximumLogValue=1.1, LogBoundary='Left')
FilterEvents(InputWorkspace='test31', SplitterWorkspace='filter', OutputWorkspaceBaseName='testtest',
             OutputWorkspaceNames='testtest_unfiltered,testtest_0')

# Compare the CurrentSP log to the one from this workspace
FilterByLogValue(InputWorkspace='test31', OutputWorkspace='old', LogName='SF1',
                 MinimumValue=0.9, MaximumValue=1.1, LogBoundary='Left')

```

### Expected behavior

### Actual behavior

### Steps to reproduce the behavior

### Platforms affected

NickDraper, 2018-03-01:
Is this essential for this release? or can we move it to R3.13?

mdoucet, 2018-03-01:
This bug affects an instrument we are currently upgrading. I have a temporary solution based on `FilterByLogValue` that I'd like to change to use `FilterEvents`. But since I expect that the instrument will run on nightly for a while, putting the fix in R3.12 is not essential unless it affects other instruments adversely.

wdzhou, 2018-06-05:
CurrentSP: ['2017-11-10T03:12:06.342000000' '2017-11-10T03:12:31.607000000'
 '2017-11-10T03:12:40.732000000']

Slicer: 
2017-11-10T03:13:06.814538624  2017-11-10T03:14:07.764311936 
2017-11-10T03:15:07.697312000  2017-11-10T03:16:08.827971840 
2017-11-10T03:17:08.745746688  2017-11-10T03:18:09.676174208 
2017-11-10T03:19:09.793680000  2017-11-10T03:20:10.757950208 
2017-11-10T03:21:10.856704640  2017-11-10T03:22:11.806604672 
2017-11-10T03:23:11.738126592  2017-11-10T03:24:12.871164928 
2017-11-10T03:25:12.786185856  2017-11-10T03:26:13.734739712 
2017-11-10T03:27:13.817295616  2017-11-10T03:28:14.800110208 
2017-11-10T03:29:14.698290816  2017-11-10T03:30:15.847703424 
2017-11-10T03:31:15.747175936  2017-11-10T03:32:16.695303808 
2017-11-10T03:33:16.810746880  2017-11-10T03:34:17.760263808 
2017-11-10T03:35:17.675607424  2017-11-10T03:36:39.758055808 
2017-11-10T03:37:38.839628416  2017-11-10T03:38:38.788989440 
2017-11-10T03:39:38.671121920  2017-11-10T03:40:39.836855936 
2017-11-10T03:41:39.737557504  2017-11-10T03:42:40.690636672

PR #22546 (2018-06-15): Fix an issue with zero entry in split TimeSeriesProperty
Method splitByTimeVector may generate split TimeSeriesProperty with zero entry.  The behaviour shall be changed such that there shall be at least one entry in each split TimeSeriesProperty instance.

**Report to:** [mdoucet ]. 

**To test:**

```
LoadEventNexus(Filename='/SNS/REF_M/shared/ADARA.Test.Data.2018/REF_M_28131.nxs.h5',
               OutputWorkspace='test31')
               
GenerateEventsFilter(InputWorkspace='test31', OutputWorkspace='filter', InformationWorkspace='filter_info', LogName='SF1',
                     MinimumLog [... cut]
closed 2018-06-15 by quantumsteve