mantidproject/mantid #21855: Unscripted Testing TofConverter

keeeto, 2018-02-26:
1. Guidelines for unscripted testing are available here: http://www.mantidproject.org/Unscripted_Manual_Testing
2. Comment against this ticket the OS environment you are testing against.
3. Don't spend more than a few hours on the testing; fatigue will kick in!
4. If you find errors in the (possibly scant) documentation, please correct them.
5. Close the this issue once you are done.
If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

Mantid-Matthew, 2018-02-27:
Testing on Ubuntu 16.04

Mantid-Matthew, 2018-02-28:
This all seems to work still.

closed 2018-02-28 by Mantid-Matthew