mantidproject/mantid #21859: Unscripted Testing FilterEvents

keeeto, 2018-02-26:
1. Guidelines for unscripted testing are available here: http://www.mantidproject.org/Unscripted_Manual_Testing
2. Comment against this ticket the OS environment you are testing against.
3. Don't spend more than a few hours on the testing; fatigue will kick in!
4. If you find errors in the (possibly scant) documentation, please correct them.
5. Close the this issue once you are done.
If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

SimonHeybrock, 2018-03-05:
Cannot plot a sample log:

```Traceback (most recent call last):
  File "/opt/mantidnightly/scripts/FilterEvents/eventFilterGUI.py", line 710, in plotLogValue
    t0ns = t0.totalNanoseconds()
AttributeError: 'numpy.datetime64' object has no attribute 'totalNanoseconds'```

SimonHeybrock, 2018-03-05:
```Traceback (most recent call last):
  File "/opt/mantidnightly/scripts/FilterEvents/eventFilterGUI.py", line 710, in plotLogValue
    t0ns = t0.totalNanoseconds()
AttributeError: 'numpy.datetime64' object has no attribute 'totalNanoseconds'
```

SimonHeybrock, 2018-03-05:
```
GenerateEventsFilter-[Error] Error in execution of algorithm GenerateEventsFilter:
GenerateEventsFilter-[Error] Fatal Error: Input minimum log value 416.3 is larger than maximum log value 279.987
```

SimonHeybrock, 2018-03-05:
Apart from the two issues linked above I have not found further issues. I will be on leave for two weeks starting tomorrow, so I am reassigning this to Owen.

@OwenArnold Thanks for picking this up. In principle the only thing left to do is to make sure that #21935 is solved before the release. As far as I can tell the other issue is not a high priority (probably not new).

NickDraper, 2018-03-05:
#21935 is tracked in the R3.12 project, so this issue can close

closed 2018-03-05 by NickDraper