mantidproject/mantid #21860: Unscripted Testing Vates

keeeto, 2018-02-26:
1. Guidelines for unscripted testing are available here: http://www.mantidproject.org/Unscripted_Manual_Testing
2. Comment against this ticket the OS environment you are testing against.
3. Don't spend more than a few hours on the testing; fatigue will kick in!
4. If you find errors in the (possibly scant) documentation, please correct them.
5. Close the this issue once you are done.
If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

Including data loading in Vates via ParaView. For help see [[Testing_Notes_2.5.0|here]]

LamarMoore, 2018-03-09:
Tested on Windows 10 and Centos 7. Vates works fine. Need to ensure that paraview can load files produced by Mantid.

OwenArnold, 2018-03-09:
Tested the SliceViewer and associated tools (live viewer, peaks viewer) on windows. Only one minor issue found, which is not new, and very minor, will report separately.

LamarMoore, 2018-03-09:
* On Centos 7, segfault when trying to load a Mantid MD workspace as a nexus file see issue [here]*
(https://github.com/mantidproject/mantid/issues/22019)
* Tested on 2 windows 10 laptops - Paraview fails to load all Mantid plugins after setting `PV_PLUGIN_PATH` correctly. Unable to continue with test for whether or not the data is loaded into paraview on Windows for this reason. See issue [here](https://github.com/mantidproject/mantid/issues/new)

closed 2018-03-13 by NickDraper