mantidproject/mantid #21861: Unscripted Testing Engineering Diffraction

keeeto, 2018-02-26:
1. Guidelines for unscripted testing are available here: http://www.mantidproject.org/Unscripted_Manual_Testing
2. Comment against this ticket the OS environment you are testing against.
3. Don't spend more than a few hours on the testing; fatigue will kick in!
4. If you find errors in the (possibly scant) documentation, please correct them.
5. Close the this issue once you are done.
If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

ElliotAOram, 2018-03-02:
docs for diffraction testing require an update:
1. Files don't run correctly:
```
Error in execution of algorithm EnggFitPeaks:
Could not fit any peak.  Failed to fit peaks with peak type BackToBackExponential even though FindPeaks found 28 peaks in principle. See the logs for further details.
  at line 97 in 'C:/MantidInstall/plugins/python/algorithms\EnggFitPeaks.py'
  caused by line 230 in 'C:/MantidInstall/plugins/python/algorithms\EnggFitPeaks.py'
Error in execution of algorithm EnggCalibrate
Could not fit any peak.  Failed to fit peaks with peak type BackToBackExponential even though FindPeaks found 28 peaks in principle. See the logs for further details.
  at line 97 in 'C:/MantidInstall/plugins/python/algorithms\EnggFitPeaks.py'
  caused by line 230 in 'C:/MantidInstall/plugins/python/algorithms\EnggFitPeaks.py'
  at line 137 in 'C:/MantidInstall/plugins/python/algorithms\EnggCalibrate.py'
  caused by line 171 in 'C:/MantidInstall/plugins/python/algorithms\EnggCalibrate.py'
Error in calibration. The calibration calculations failed. One of the algorithms did not execute correctly. See log messages for further details. Error: EnggCalibrate failed
```
2.  Lacks some of the new tabs (GSAS etc.)

ElliotAOram, 2018-03-13:
All interfaces here seem fine except for the GSAS GUI which is not working in the installer. Joe is currently looking into this.
This can now be closed as all relevant issues have been found

closed 2018-03-13 by ElliotAOram