mantidproject/mantid #21864: Unscripted Testing ISIS Reflectometry (old and new)

keeeto, 2018-02-26:
1. Guidelines for unscripted testing are available here: http://www.mantidproject.org/Unscripted_Manual_Testing
2. Comment against this ticket the OS environment you are testing against.
3. Don't spend more than a few hours on the testing; fatigue will kick in!
4. If you find errors in the (possibly scant) documentation, please correct them.
5. Close the this issue once you are done.
If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

For testing procedures see http://www.mantidproject.org/Testing_ISIS_Reflectometry_GUI

Mantid-Matthew, 2018-02-28:
Testing on Ubuntu 16.04

keeeto, 2018-03-09:
Testing on OSX 10.12

keeeto, 2018-03-09:
Issues around plotting.

* Before rows are processed the plot functions should be greyed out. They are not. However no plot can be generated

* After processing a row and then editing the data the plot buttons should be greyed out. They are not. Also it is still possible to plot, the old data is used, this should not happen.

I will create issues related to this.

keeeto, 2018-03-09:
I am having trouble getting "Search for an RB Number" to work on Mac OSX. Not sure if this is an issue with my setup or with the code. I will continue to test the rest of the features.

keeeto, 2018-03-09:
Created issue #22034 about the plotting buttons.

keeeto, 2018-03-12:
Created issue #22050 about copy and paste behaviour.

keeeto, 2018-03-12:
Created issue #22051 about lack of a save/discard/cancel propmt when closing the interface after editing data.

keeeto, 2018-03-12:
This testing is now finished. Note that the [Reflectometry Testing documentation](https://www.mantidproject.org/Testing_ISIS_Reflectometry_GUI#Reflectometry_New) has now been updated to include different instructions for the `Old` and `New` interfaces. The testing procedures should probably be reviewed by someone closer to the development to make sure they are comprehensive enough.

The following issues were created from the testing:
* Plot buttons in reflectometry interface not behaving as expected #22034
* Incorrect copy and paste behaviour in ISIS Reflectometry #22050
* No prompt to save changes on closing ISIS Reflectometry #22051

closed 2018-03-12 by keeeto