mantidproject/mantid #21866: Unscripted Testing Live Data ISIS

keeeto, 2018-02-26:
1. Guidelines for unscripted testing are available here: http://www.mantidproject.org/Unscripted_Manual_Testing
2. Comment against this ticket the OS environment you are testing against.
3. Don't spend more than a few hours on the testing; fatigue will kick in!
4. If you find errors in the (possibly scant) documentation, please correct them.
5. Close the this issue once you are done.
If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

gemmaguest, 2018-03-06:
Testing on Windows 7

gemmaguest, 2018-03-07:
Ran live data for around an hour with live offspec data. Tested with/without postprocessing algorithm and/or script. Ran with slice viewer, colour plot, spectrum viewer and mantid matrix all open on the updating workspaces without problem. Running multiple live data monitor algorithms and cancelling them all worked ok. Mantid used quite a lot of memory while running but got back down to around 400MB after exiting live data and removing all workspaces.

Most things also ran without problem for the fake instruments (event & histo). However, I left FakeISISHistoDAE and MonitorLiveData running overnight with slice viewer open, and Mantid was using 4GB in the morning. After cancelling the algorithms and removing all workspaces it was still using 3GB.

gemmaguest, 2018-03-12:
As far as I can tell the potential leak doesn't look like anything to do with this release and seems more likely to be to do with the post processing algorithm rather than live data anyway. I'll therefore close this issue.

closed 2018-03-12 by gemmaguest