mantidproject/mantid #21867: Manual Testing - Mantid Basics Course

keeeto, 2018-02-26:
1. Guidelines for unscripted testing are available here: http://www.mantidproject.org/Unscripted_Manual_Testing
2. Comment against this ticket the OS environment you are testing against.
3. Don't spend more than a few hours on the testing; fatigue will kick in!
4. If you find errors in the (possibly scant) documentation, please correct them.
5. Close the this issue once you are done.
If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

See https://www.mantidproject.org/Mantid_Basic_Course

NickDraper, 2018-03-02:
raised issues:
https://github.com/mantidproject/mantid/issues/21929

NickDraper, 2018-03-02:
passed other than that, updated a couple of pages in the training

closed 2018-03-02 by NickDraper