mantidproject/mantid #21876: Slow builds of user docs

peterfpeterson, 2018-02-26:
#21847 slowed down the builds of user documentation significantly. On my system the change is 2m30s goes up to 9m30s. The way to test it is to modify `get_file_last_modified` in `sourcelink.py` to be
```python
    def get_file_last_modified(self, filename):
        return LAST_MODIFIED_UNKNOWN
```
Then `rm docs/doctrees/algorithms/* ; time ninja docs-html`.

The new functionality is quite nice, but the slowdown isn't helpful. A way would be to get the whole history in one shell call then parse the result for what you need. The closest I found for that so far is
```sh
$ git log --pretty=format:"%cd" --date=short  --name-only Framework/
```
I don't know if it matters, but there should probably be a check that this is in a git repository too.

martyngigg, 2018-02-27:
The single-shot call is similar to what is done when finding the file paths. The first time it caches the results for everything: https://github.com/mantidproject/mantid/blob/master/docs/sphinxext/mantiddoc/directives/sourcelink.py#L124

peterfpeterson, 2018-02-27:
You pay the full time penalty anytime sphinx configuration or plugins change.

DanNixon, 2018-03-01:
@peterfpeterson Are you sure this slowdown is solely the fault of #21847?

I tired to reproduce this as you mentioned, but I removed the entire `docs` directory and it still takes >9 minutes to build for me.
Building only the algorithm pages (by removing `docs/doctrees/algorithms/*`) shows only a minute increase in build time (which I'll have a PR to attempt to reduce shortly).

peterfpeterson, 2018-03-01:
I'm curious why the difference is much smaller for you. The reason why I think it is the "last modified" code is because having it return a dummy string rather than asking git was so very much faster and reproducible (with smaller than 10 second difference between attempts).

DanNixon, 2018-03-01:
Were you rebuilding the docs from clean every time? In your original steps you only remove the algorithm pages, so if something was causing a slow down outside of those docs then you wouldn't be measuring it.

I arrived at around a minute increase by:
- building the full user docs (which takes >9 minutes)
- removing algorithm pages and modifying `get_file_last_modified` as per your original steps
- building docs (which now takes around 2 minutes)
- removing algorithm pages and revert the change to `get_file_last_modified`
- building docs (which takes around 3 minutes)

peterfpeterson, 2018-03-01:
The test to do is
1. build the docs (ignoring time)
2. remove `docs/doctrees/algorithms/*` and build (noting time)
3. change `get_file_last_modified` and build (ignoring time)
4. remove `docs/doctrees/algorithms/*` and build (noting time)
I'll do it a few more times and add the statistics here.

peterfpeterson, 2018-03-01:
On my fc26 system:
* existing implementation takes 9m58s +- 16s
* changing `get_file_last_modified` to return a string without doing anything takes 2m34s +- 8s

DanNixon, 2018-03-02:
I can only imagine the issue is trying to run this on hardware with slow disk IO.

You could try using the cacheing method (https://github.com/mantidproject/mantid/compare/master...DanNixon:21876_cache_commit_dates_for_docs) but I doubt it will help the situation (for me it makes it around 30 seconds slower).
Ideally you need the list of files you care about up front, but I can't see any way of getting that with Sphinx.

If it doesn't help then I vote to just abandon the timestamps, at the end of that day it is only saving a single click (i.e. following the link and getting the GitHub page for the file in question).

PR #22173 (2018-03-22): Add cacheing of Git commit timestamps in doc build
Fixes #21876

Caches Git commit timestamps, reducing time taken to build user docs.

**To test:**

- Build user docs and measure time
- Compare that to time taken for current master

**Release Notes** 

*Does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the code conform to the [coding standards](http://www.mantidproject.org/Coding_Standa [... cut]
closed 2018-03-22 by martyngigg