mantidproject/mantid #21884: ALC background

AnthonyLim23, 2018-02-27:
In the muon ALC interface the background is restricted to have x > 0, this should not be the case

AnthonyLim23, 2018-03-01:
Set HIFI as default instrument. Open the interface. Put First run = 127250 and Last = 127280. Don't tick Auto.
Select Log = sample_magn_field, take at Mean.
Select Dead times from file (this probably doesn't make any difference to the bug). Leave grouping, etc at default (again shouldn't make much difference).
Load.
You should have a sharp resonance line on a slightly sloping background, with an x range between -150 and +150.

Go on to "Baseline modelling".
Right click in the top left pane and choose function LinearBackground.
Right click in the bottom left pane and do Add Section. It'll put two vertical dot-dashed lines, one at about x=0 and one at x=150.
Try either to drag the left line past x=0 into the negative region, or to type negative numbers into the table on the left. Ideally I'd like one section from x=-150 to x=-100 and another from +100 to +150.

martyngigg, 2018-03-06:
Closed when merging #21947

PR #21947 (2018-03-06): Allow negative sections in ALC background
Changed the initial range when the selector is created from dbl_min (near 0) and dbl_max(huge).  The max and min were set later and when positive overrode the range, as they were within it.  Negative values were ignored for the min as they were not within the range.

**To test:**

Set HIFI as default instrument. Open the interface. Put First run = 127250 and Last = 127280. Don't tick Auto.
Select Log = sample_magn_field, take at Mean.
Select Dead times from file (this probably doesn't make any difference to the bug). Leave grouping, etc at default (aga [... cut]
closed 2018-03-06 by martyngigg