mantidproject/mantid #21893: Unscripted Testing ISIS Indirect interfaces

keeeto, 2018-02-27:
1. Guidelines for unscripted testing are available here: http://www.mantidproject.org/Unscripted_Manual_Testing
2. Comment against this ticket the OS environment you are testing against.
3. Don't spend more than a few hours on the testing; fatigue will kick in!
4. If you find errors in the (possibly scant) documentation, please correct them.
5. Close the this issue once you are done.
If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

See
* http://www.mantidproject.org/Indirect:ConvertToEnergy
* http://www.mantidproject.org/Indirect:Indirect_Data_Analysis
* http://www.mantidproject.org/UsingElwin
* http://www.mantidproject.org/UsingFury

NickDraper, 2018-03-01:
As Jo has gone on Holiday, I've transferred this to you for now

ElliotAOram, 2018-03-02:
Low priority:
*ISIS Indirect Data Reduction* - Plot time for multiple spectra (eg. SpectraMin=3, SpectraMax=5) curve in graph is just labeled as `<run_number>_grp`. Would be better to have this actually display the run numbers

ElliotAOram, 2018-03-06:
It's not obvious what (if anything) `Plot Raw` in Indirect Data reduction > ISIS Calibration does. When clicked nothing appears to happen.

ElliotAOram, 2018-03-06:
Bayes
====
**ResNorm**
👍 
**Quasi**
👍 
**Stretch** 
* If you increase the `Current Spectrum` spinner and then add a workspace with fewer spectra than the `Current Spectrum` spinner you get an unexpected exception:
  * `Unexpected exception: Workspace index is out of range, cannot plot.`

ElliotAOram, 2018-03-07:
Corrections
====
**Container Subtraction**
* `Plot Current Preview` button plots the current workspace indexes from the loaded data but does not apply the shift and scale. This should plot the same as the preview plot, or the the button name should be changed to a better description of what it does.
  * Additionally, plot current preview does not plot the subtracted data either 

**CalculatePaalmanPings**
* Nit: Can have 0 mass density in Sample/Can details. You are provided with a warning in the log about this but it is fairly obscured by other output in default log level - do we want to be more obvious about this error? Answer could be no depending on use case

**Calc Monte Carlo Abs** 
* Same as above but this time there is no warning at all when mass density is 0. We should at the bare minimum inform users about what happens in this case?

**Apply Abs Corr*
* Same plotting issue as `Container Subtraction`

ElliotAOram, 2018-03-07:
Data Analysis
=========
**Elwin**
* Future improvement: Can't take workspace input only file (unlike the majority of other interfaces)

**MSD Fit**
* High priority: `Unexpected exception: Invalid value for property Model (string) "None": The value "None" is not in the list of allowed values` is produced hen clicking `Fit Single Spectrum` with the following input:
[image]
* Future improvement : Allow drag and drop capability for fit function menu rather than only double click docking/undocking.

**I(Q, t)**
* High priority: If you drag the range bars to a near zero number you get the following exception: `Error in execution of algorithm TransformToIqt:
float division by zero
  at line 62 in 'C:/MantidInstall/plugins/python/algorithms\WorkflowAlgorithms\TransformToIqt.py'
  caused by line 133 in 'C:/MantidInstall/plugins/python/algorithms\WorkflowAlgorithms\TransformToIqt.py'`

**I(Q, t) Fit**
* High priority: Using the `iris26176_graphite002_iqt` data set produced from the `I(Q, t)` tab, if you run `Setup>Find peaks` without any other options, Mantid crashes - note this is not caught by the new error reporting?
* Can't drag range bars in top preview plot
* High Priority: Removing the fitting parameters (can be as simple as adding/removing a function) causes mantid to crash when the function is removed.
* Fitting stretch on the `iris 26176 iqt` gets stuck and crashes mantid?

ElliotAOram, 2018-03-07:
Test data used:
[isis-indirect-test-files.zip](https://github.com/mantidproject/mantid/files/1789394/isis-indirect-test-files.zip)

closed 2018-03-08 by ElliotAOram