mantidproject/mantid #21905: MSD Fit in Indirect Data Analysis crashes when attempting to fit

josephframsay, 2018-02-28:
### Expected behavior

The MSD fit tab of the IDA GUI crashes when **Run** is clicked for a workspace (or nxs file) created using the Elwin tab (such as [here](https://www.mantidproject.org/Indirect:Indirect_Data_Analysis#Running_Elwin). The error message is `Unexpected exception: Invalid value for property Model (string) "None": The value "None" is not in the list of allowed values`

### Actual behavior

Fitting should be run on the workspace

### Steps to reproduce the behavior

Follow the tutorial [here](https://www.mantidproject.org/Indirect:Indirect_Data_Analysis). The **MSD Fit** step should cause a crash

SpencerHowells, 2018-03-01:
The whole page is years out-of-date. For MSDfit the file now required is _eq not _eq2.

brandonhewer, 2018-03-22:
No longer applicable

closed 2018-03-22 by brandonhewer