mantidproject/mantid #21906: Indirect Data Analysis ConvFit tab crash, no documentation

josephframsay, 2018-02-28:
The documentation for the ConvFit tab ([here](https://www.mantidproject.org/Indirect:Indirect_Data_Analysis)) is no longer relevant to the GUI, and as a result it is not clear how to actually run a fit. If you load in the suggested workspaces from the ISIS sample data, the click Fit -> Fit, MantidPlot crashes with the following exception:
`Unexpected exception: Invalid value for property Workspace (Workspace) "irs26176_graphite002_conv_FixF_s0_Workspaces": Workspace "irs26176_graphite002_conv_FixF_s0_Workspaces" was not found in the Analysis Data Service
`

brandonhewer, 2018-03-05:
@josephframsay It sounds as though you run the ConvFit tab without a Fit Function selected, although this is no longer clear from the Documentation, which is several years out of date. 
Of course, it shouldn't be possible to crash MANTID - it may be best to separate this issue into two parts -- one for documentation and one for this crashing error -- since the crashing problem also would appear to be the same problem outlined in #21905

robertapplin, 2018-08-31:
This is no longer an issue (as of Mantid version 3.13)

closed 2018-08-31 by robertapplin