mantidproject/mantid #21935: Cannot plot log value in Filter-Events GUI

SimonHeybrock, 2018-03-05:
In the `Filter Events` interface:
1. Load file (e.g, `CNCS_7860_event.nxs` from the traning course data), click `Use`.
2. Select an item in `Sample Log` drop down menu, click `Plot`:

```Traceback (most recent call last):
  File "/opt/mantidnightly/scripts/FilterEvents/eventFilterGUI.py", line 710, in plotLogValue
    t0ns = t0.totalNanoseconds()
AttributeError: 'numpy.datetime64' object has no attribute 'totalNanoseconds'```

OS: Ubuntu 16.04.

NickDraper, 2018-03-05:
@peterfpeterson Is ther someone in the US that can take a look at this?

peterfpeterson, 2018-03-05:
@AndreiSavici is this something that could be "easily" fixed using the new log-plotting code that just got merged in?

PR #21951 (2018-03-06): Change conversion to seconds
`np.datetime64` has better methods than mantid's `DateAndTime` object so this uses those.

**To test:**

Try out the test from the original issue. 

Fixes #21935.

*Does not need to be in the release notes* because it fixes a bug introduced during this release cycle.
---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the code conform to the [coding standards](http://www.mantidproject.org/Coding_Standard [... cut]
closed 2018-03-06 by NickDraper