mantidproject/mantid #21937: Pearl Focus runs issue

ElliotAOram, 2018-03-05:
Nick Funnell reports a difference between the old PEARL focusing routines vs the new ones implemented in ISIS_Powder.
The initial estimation of the origin of the difference is the vanadium normalisation.
The changes made in this issue will most likely involve updating the system tests

NickDraper, 2018-03-12:
Waiting for input from Nick Funnell

ElliotAOram, 2018-03-13:
This is believed to be an input issue rather than something that we are required to change. @josephframsay is checking this with Nick Funnell and will close this issue if this is the case.

josephframsay, 2018-03-14:
As far as I can tell, this issue came up because absorption corrections had not been applied to the vanadium. Will reopen if the problem persists for Nick, but I think this is probably a non-issue

closed 2018-03-14 by josephframsay