mantidproject/mantid #21943: CrystalFieldPythonInterface system test crashes on Windows 7

martyngigg, 2018-03-05:
The [master system tests](http://builds.mantidproject.org/view/Master%20Pipeline/job/master_systemtests-win7/568/testReport/junit/SystemTests/CrystalFieldPythonInterface/CrystalFieldPythonInterface/) are showing that the new `CrystalField.PythonInterface` test has started failing but only on Windows 7 and not 10.

mducle, 2018-03-05:
@martyngigg @gemmaguest So, it bombs out on [line 295](https://github.com/mantidproject/mantid/blob/master/Testing/SystemTests/tests/analysis/CrystalFieldPythonInterface.py#L295) of the test, which is a `fit` command but the input dataset does not agree with the input model (it was generated for a different crystal field model). 

This means that the fit will never converge. In other systems the `fit` algorithm just seems to run until it reaches the maximum iteration and moves on, but on Windows 7, for some reason it crashes. 

If you add

```
ws = makeWorkspace(*cf.getSpectrum())
```

before [line 292](https://github.com/mantidproject/mantid/blob/master/Testing/SystemTests/tests/analysis/CrystalFieldPythonInterface.py#L292) where the fit is setup, then the fit will now converge and the system test should run ok (it does when I run it with the build-server built exec). 

However, I don't know why we're getting a memory error on that particular build and not others. Also, I've run it a couple of times with the build-server exec without the above additional line and sometimes it actually runs - the `fit` takes about 2-3min to decide it can't fit, and uses about 800MB of memory to do so but it runs; other times it crashes. I think there's a memory leak somewhere but not sure whether it's in Mantid or `gsl` [VS says the memory error is in `gsl.dll`] or how to debug it on Windows. Does the Win7 / Win10 build servers have the same versions of `gsl`?

martyngigg, 2018-03-06:
@mducle Thanks for looking into this further.

This is quite odd behaviour. The gsl version is identical across all windows builds and we haven't changed it lately. It's also odd that it behaves differently across different runs. This is going to be difficult to debug.

martyngigg, 2018-03-06:
I'm going to try it directly on the copy that the build server built to see if I can spot any differences.

martyngigg, 2018-03-06:
I'm beginning to think this could be something to do with the machine it was built on. I've configure it so the build will happen somewhere else tonight. Let's see where we get to with that.

martyngigg, 2018-03-07:
After restarting the Windows machine yesterday the builds now seem to be passing. This is on both the [`master`](http://builds.mantidproject.org/view/Master%20Pipeline/) and [`release`](http://builds.mantidproject.org/view/Release%20Pipeline/) pipelines where the installers were built on different machines. I did restart the machines yesterday as there were some updates pending so I think we'll have to put this down to something strange about the `isis-ndw1185` machine that a restart as "cured".

Please reopen this if anyone thinks we should look into it further.

martyngigg, 2018-03-09:
This still appears to be an issue on both release and master branches. Opening to continue investigation.

martyngigg, 2018-03-09:
I've tracked it so far as a call to `gsl_set_matrix` that seems to receive a garbage value as the column index to set but it's not clear what is causing it.

mducle, 2018-03-12:
@martyngigg I think what happened part-way through the fit is that the number of peaks in the `CrystalFieldSpectrum` composite function increased because the parameters being varied in the underlying `CrystalFieldPeaks` function changed enough to put more peaks into the intensity and energy range of the fit. 

We have two thresholds for which peaks are included - `ToleranceIntensity` and `ToleranceEnergy`. `ToleranceIntensity` removes the very low intensity peaks, and `ToleranceEnergy` removes degenerate peaks with nearly the same energy. In this case I would guess that it is `ToleranceEnergy` because of the way the point charge model calculation (which gives the initial parameters in this case) is set up. The actual crystal structure used here (of `Sm2O3`) has quite a high symmetry, which means that the energy levels will be quite degenerate, so the initial parameters will give only a few peaks. 

Now, the crystal field parameters have to be defined relative to a direction in the crystal, but the point charge function doesn't know what the high symmetry directions are. So, it just choses an arbitrary direction. However, this means that the parameters it outputs, **when varied by the fit** will give lower symmetry crystal field splittings where the energy levels will no longer be degenerate - e.g. it will increase the number of peaks.

In any case, there is a maximum number of peaks in the code, `maxNPeaks`, and I thought there was a check that ignores any peaks above this (the peaks are sorted by intensity, so it would ignore the lowest intensity peaks). I'm not sure what went wrong here. @mantid-roman any idea?

Another solution for this particular case (of the point charge model) is for the point charge code to set ties on the parameters which would mean that the crystal symmetry, and hence the energy level degeneracies, are respected. I'm not quite sure how to calculate what the ties should be at the moment, though, so this would be a longer term solution.

mantid-roman, 2018-03-12:
I don't think `maxNPeaks` limits the number of fitting parameters if we allow to fit the widths. And the GSL minimizers need to know the number of actual free parameters. But `Fit` is supposed to deal with it by re-initialising the minimizer when the number of parameters changes. But apparently something went wrong here.

PR #22047 (2018-03-13): Fix access violation in CrystalField.PythonInterface test
Description of work.

Checks whether the number of active parameters changes during function evaluation and throws if it does.

**To test:**

* Code review
* On Windows run the `CrystalField.PythonInterface` test using a release build and there should be no access violation.
* A full check would be to run `valgrind` again too.

Fixes #21943

**Release Notes** 
*This should be transparent to users so it does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mant [... cut]
closed 2018-03-13 by NickDraper