mantidproject/mantid #21955: Segmentation fault when `Execute Selection` in script window using `LoadILLReflectometry`

reimundILL, 2018-03-06:
Mantid crashes with segmentation fault when running the following script in the Mantid script window and trying then to `Execute Selection` (e.g. selecting the first line)

```Python
db = "/ExternalData/Testing/Data/UnitTest/ILL/Figaro/592724.nxs"  
DB = LoadILLReflectometry(Filename=db, XUnit='TimeOfFlight')
sanDB = DB.getRun().getProperty('SAN.value')
DB = LoadILLReflectometry(Filename=db, XUnit='TimeOfFlight', BraggAngle=sanDB.value)
```

It seems to only crash with the `BraggAngle` option that should be the easiest (avoids fitting).
It seems to only crash when selecting `Execute` -> `Execute Selection` in the script window.
Tested on master.
Ubuntu 16.04.

reimundILL, 2018-03-06:
Some further observations:
- no crash with debug build

reimundILL, 2018-03-06:
********* SEGMENTATION FAULT *********
 /users/reimund/mantid/bin/libMantidAPI.so(+0x346849) [0x7f122ab14849]
 /users/reimund/mantid/bin/libMantidAPI.so(+0x346a78) [0x7f122ab14a78]
 /lib/x86_64-linux-gnu/libc.so.6(+0x354b0) [0x7f1222fc24b0]
 /usr/lib/x86_64-linux-gnu/libQtCore.so.4(_ZNK18QThreadStorageData3getEv+0x1b) [0x7f1226877eeb]
 /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x4ce451) [0x7f1227406451]
 /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x4ce586) [0x7f1227406586]
 /usr/lib/x86_64-linux-gnu/libQtGui.so.4(_ZNK13QFontEngineFT9loadGlyphEPNS_9QGlyphSetEj6QFixedN11QFontEngine11GlyphFormatEb+0x1e9) [0x7f1227408e79]
 /usr/lib/x86_64-linux-gnu/libQtGui.so.4(_ZNK13QFontEngineFT14recalcAdvancesEP12QGlyphLayout6QFlagsIN11QTextEngine10ShaperFlagEE+0x105) [0x7f1227409fd5]
 /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x3f9fb1) [0x7f1227331fb1]
 /usr/lib/x86_64-linux-gnu/libQtCore.so.4(+0xeeed9) [0x7f12268eded9]
 /usr/lib/x86_64-linux-gnu/libQtCore.so.4(+0xf0df4) [0x7f12268efdf4]
 /usr/lib/x86_64-linux-gnu/libQtCore.so.4(+0xf5c73) [0x7f12268f4c73]
 /usr/lib/x86_64-linux-gnu/libQtGui.so.4(_ZNK11QTextEngine21shapeTextWithHarfbuzzEi+0x7e7) [0x7f1227365567]
 /usr/lib/x86_64-linux-gnu/libQtGui.so.4(_ZNK11QTextEngine9shapeTextEi+0x9a) [0x7f1227365efa]
 /usr/lib/x86_64-linux-gnu/libQtGui.so.4(_ZNK11QTextEngine5shapeEi+0xa5) [0x7f1227366275]
 /usr/lib/x86_64-linux-gnu/libQtGui.so.4(_ZN9QTextLine13layout_helperEi+0x10a1) [0x7f1227376cc1]
 /usr/lib/x86_64-linux-gnu/libQtGui.so.4(_ZN11QTextLayout9endLayoutEv+0x85) [0x7f1227377a55]
 /usr/lib/libqscintilla2.so.12(_ZN11SurfaceImpl13MeasureWidthsER4FontPKciPf+0xd1) [0x7f1225eef9ad]
 /usr/lib/libqscintilla2.so.12(_ZN13PositionCache13MeasureWidthsEP7SurfaceRK9ViewStylejPKcjPfP8Document+0x365) [0x7f1226018a73]
 /usr/lib/libqscintilla2.so.12(_ZN8EditView10LayoutLineERK9EditModeliP7SurfaceRK9ViewStyleP10LineLayouti+0xcb0) [0x7f1225ffe52a]
 /usr/lib/libqscintilla2.so.12(_ZN8EditView9PaintTextEP7SurfaceRK9EditModel10PRectangleS5_RK9ViewStyle+0x71e) [0x7f1226007488]
 /usr/lib/libqscintilla2.so.12(_ZN6Editor5PaintEP7Surface10PRectangle+0x446) [0x7f1225fdcde6]
 /usr/lib/libqscintilla2.so.12(_ZN15QsciScintillaQt10paintEventEP11QPaintEvent+0x1fa) [0x7f1225ef3cfc]
 /usr/lib/libqscintilla2.so.12(_ZN17QsciScintillaBase10paintEventEP11QPaintEvent+0x27) [0x7f1225eb9823]
 /usr/lib/x86_64-linux-gnu/libQtGui.so.4(_ZN7QWidget5eventEP6QEvent+0x2a0) [0x7f12271538d0]
 /usr/lib/x86_64-linux-gnu/libQtGui.so.4(_ZN6QFrame5eventEP6QEvent+0x1e) [0x7f122751ccee]
 /usr/lib/x86_64-linux-gnu/libQtCore.so.4(_ZN23QCoreApplicationPrivate29sendThroughObjectEventFiltersEP7QObjectP6QEvent+0x86) [0x7f1226989a76]
 /usr/lib/x86_64-linux-gnu/libQtGui.so.4(_ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent+0x6c) [0x7f12270fcfbc]
 /usr/lib/x86_64-linux-gnu/libQtGui.so.4(_ZN12QApplication6notifyEP7QObjectP6QEvent+0x2b6) [0x7f1227103f16]
 ./bin/MantidPlot() [0x839e6f]
 /usr/lib/x86_64-linux-gnu/libQtCore.so.4(_ZN16QCoreApplication14notifyInternalEP7QObjectP6QEvent+0x8d) [0x7f122698990d]
 /usr/lib/x86_64-linux-gnu/libQtGui.so.4(_ZN14QWidgetPrivate10drawWidgetEP12QPaintDeviceRK7QRegionRK6QPointiP8QPainterP19QWidgetBackingStore+0x607) [0x7f122714df37]
Segmentation fault

reimundILL, 2018-03-06:
Mantid does not crash when changing the name of the second OutputWorkspace -> a problem exists if the workspace already exists in the ADS and needs to be replaced by the newly loaded workspace.

reimundILL, 2018-03-06:
Mantid does not crash if not using line 3 to get the san angle from run of the workspace but giving a number instead.

reimundILL, 2018-03-06:
More general:

This script always crashes Mantid when executing:

```Python
A = CreateSampleWorkspace()
B = A.getRun().getProperty('run_title')
A = CreateSampleWorkspace()
A.setTitle(B)
```

martyngigg, 2018-03-06:
@reimundILL Thanks for spotting this. The issue is that `getProperty` returns a pointer to an object held in workspace `A`. After running the second `CreateSampleworkspace` `B` becomes a dangling pointer and then seg faults when you try to use it. To address this properly we need to change how we are storing these under the hood.

In your case I think you actually want the `.value` of the property, i.e.

```
A = CreateSampleWorkspace()
B = A.getRun().getProperty('run_title').value
A = CreateSampleWorkspace()
A.setTitle(B)
```

should work as it creates a new python object for B.

reimundILL, 2018-03-06:
@martyngigg  Thanks, that makes sense. In the original script I asked something like `A.setTitle(B.value)`. There was a fear that it's the relatively new loader that causes the segfault. Even when the terminal points to libqscintilla2.

martyngigg, 2018-03-06:
@reimundILL No problem. I think the reason you are seeing `libqscintilla2` in the terminal debugger is that the script window executes the script in a separate thread. By default the command-line debugger shows the stack trace of the main thread but if you type `info threads` you can see them all. You can then switch to a different thread with `thread ID` where ID is the number given by the `info threads` command.

reimundILL, 2018-03-06:
@martyngigg Thank you, I did not know this and will try it

reimundILL, 2018-03-08:
@martyngigg  Please have a look at this little fix. It may be important for our new users.

reimundILL, 2018-03-08:
I can now execute
```Python
A = CreateSampleWorkspace()
B = A.getRun().getProperty('run_title')
A = CreateSampleWorkspace()
A.setTitle(B)
```
and get the correct error without a crash. But I can still provoke crashes with
```Python
A = CreateSampleWorkspace()
B = A.getRun().getProperty('run_title')
A = CreateSampleWorkspace()
A.setTitle(B.value)
```

martyngigg, 2018-03-08:
@reimundILL Thanks for looking into this further. I think your fix makes sense - if there has been an error in the script execution then we can't necessarily expect the auto-complete information to be up to date. I think this would be good for a fix if it were in a separate issue - would that be okay.

I also agree we have a deeper issue with your second example but that is not so simple to solve.

reimundILL, 2018-03-08:
I think I solved all crashes now, I tested both scripts, and sequential execution. Auto completion still works.

closed 2018-03-08 by reimundILL