mantidproject/mantid #22023: Crash & Muon data analysis multi fit is including, but not show auto background functions

NickDraper, 2018-03-09:
### Expected behavior
Auto background may not be desirable for muon data, so it should either be disabled, or at least shown in the GUI

### Actual behavior
If you have an auto background selection it is silently added to the function you have chosen in the muon data analysis gui

Auto background functions are listed in the single fit muon gui

### Steps to reproduce the behavior
view->preferences
Mantid->curve Fitting
set auto background = LinearBackground, click OK
Open muon analysis
Load some data (e.g. HIFI00108376)
Go to the settings tab and make sure that the multiple fitting box is ticked
Go to data analysis tab
Note that nothing is in the function list
Click Fit -> Fit
CRASH -> continue
Open the workspace HIFI00108376 .......... _ Parameters
Note the A0 and A1 parameters from the linear background function
[image]

NickDraper, 2018-03-09:
@AnthonyLim23 Another new moun issue

AnthonyLim23, 2018-03-09:
I think I know what is causing the other bug. So hope to get that fixed today or early Monday. If no one else has taken this by then I will have a look at it

AnthonyLim23, 2018-03-09:
they tend not to use auto background. So this is less important

AnthonyLim23, 2018-03-09:
make sure we clear the functions when changing interface

PR #22058 (2018-03-13): Auto background and Muon analysis
Auto background would add a function in the background of muon analysis. Doing a a multiple fit would then be possible without seeing any fitting functions and it would then be unable to update the browser and throw an error. 

**To test:**
With auto background set

Open Muon analysis
Go to the settings tab and make sure that multiple fitting is ticked 
Load some data
Go to data analysis and the fit browse should be empty
Check that the fit menu items are disabled as expected
Add a fitting function and do a fit 

Go to the settings tab and untick multiple fitt [... cut]
closed 2018-03-13 by NickDraper