mantidproject/mantid #22029: Indirect data analysis I(Q, t) Fit validation of input

ElliotAOram, 2018-03-09:
The file format checker only shows `*_iqt.nxs` files, but if you can bypass this (give full file path / select all files option in file tree). If you use the `*_red.nxs` file, check `Stretched Exponential`  and click `Run` mantid crashes to crash reporter.

Whilst this interface is not designed to be used in this way, we should still ensure that some validation stops users from seeing this.

SpencerHowells, 2018-03-09:
@ElliotAOram one possibility is to check that the workspace has x-unit as time. And perhaps also that x > 0.

ElliotAOram, 2018-03-09:
These both seem like sensible checks I'll push this to @brandonhewer for now, but let me know if you don't have enough time and I might get time next week to have a look!

SpencerHowells, 2018-03-09:
@brandonhewer x could be 0 so the check is that x < 0

SpencerHowells, 2018-03-09:
@brandonhewer wrong symbol! Check if x<0 then error

PR #22053 (2018-03-13): Indirect Data Analysis - Add I(Q,t) Fit workspace validation
An extra check has been added to the validation of workspaces for `i(q,t) Fit` interface. This checks that the first x bin is of a value greater than or equal to 0.

**To test:**

* Download this test data: 
[isis-indirect-test-files.zip](https://github.com/mantidproject/mantid/files/1803382/isis-indirect-test-files.zip)

* Open `I(Q, t) Fit` - `Interfaces > Indirect > Data Analysis > I(Q, t) Fit`
* Load the `26176_..._iqt` file from the test data
* Add a fit function (just checking the `Stretched Exponential` in the property tree wo [... cut]
closed 2018-03-13 by NickDraper