mantidproject/mantid #22034: Plot buttons in reflectometry interface not behaving as expected

keeeto, 2018-03-09:
### Expected behavior

Before data has been processed the plot buttons should be greyed out and inaccessible.

### Actual behavior

The plot buttons are always available. 
If the data has never been processed clicking plot throws an error box.
If data has been processed, then edited, but not processed again, the plot button results in a plot of the original, unedited data.

### Steps to reproduce the behavior

* Open the Reflectometry interface.
* Import a data file - `INTER_NR_test2.tbl` from the training data is good.
* Select a row and click on one of the plot buttons.
* This produces an error box.
* Now process the data.
* Click plot - this produces a plot as expected.
* Edit some of the data.
* Click plot - this produces a plot of the old data.

### Platforms affected

OSX 10.12

gemmaguest, 2018-03-12:
Greying out the buttons would be nice but is not a priority as the error message is sensible.

I'll check with the scientists about what they want to happen when settings are changed but I doubt they are too worried about this.

gemmaguest, 2020-07-20:
Closing because this is low priority and we are unlikely to get around to it. It was a developer suggestion rather than a request from users.

closed 2020-07-20 by gemmaguest

gemmaguest, 2022-03-08:
Closing as stale for now because we're not likely to get round to this