mantidproject/mantid #22050: Incorrect copy and paste behaviour in ISIS Reflectometry

keeeto, 2018-03-12:
### Expected behavior

Copying cells from excel and pasting them to a highlighted area in the interface should result in pasting as much of the original data as possible in the selected area.

### Actual behavior

The first column of data does not get pasted. 

### Steps to reproduce the behavior

* Open a new spread sheet.
* Fill n columns in a row (say n=5)
* Copy the row
* Select a row in the table in the interface
* Paste the data

### Platforms affected

OSX
Windows

gemmaguest, 2018-03-14:
The odd behaviour is because the GUI is not really designed to have copy-paste from external programs. Copy paste from within the GUI table uses the clipboard, though, which is why something sort of works here. The implementation includes 2 hidden columns, one at the start and one at the end, which is why there is the off-by-one error in what gets pasted. We could potentially fix this to make it do something sensible, but this is unlikely to be that useful so is not a priority.

More of a concern is that there is a potential out-of-bounds error due to the hidden columns, which users could hit if they accidentally paste in bad data. This has been addressed in PR #22091.

keeeto, 2018-03-14:
I agree, I was a little surprised by the presence of a test for this functionality, as I would not have expected the interface to do this by default. I'd say that, if users are not accustomed to doing copy/paste operations then there is no need to make this a priority. We could also just remove that unscripted test from the test suite for the new interface.

NickDraper, 2018-03-15:
Has the pull request solved this, in which case close this issue, or is it only a partial fix?

gemmaguest, 2018-03-15:
@NickDraper It's a partial fix

gemmaguest, 2019-05-14:
Pasting from excel is intentionally disabled in the restructured GUI so this is no longer relevant

closed 2019-05-14 by gemmaguest