mantidproject/mantid #22082: Developer documentation umbrella issue

martyngigg, 2018-03-14:
At the last developer meeting it was agreed that we would move all developer documentation to the code base. The docs will live in a new sphinx project called [`dev-docs`](https://github.com/mantidproject/mantid/tree/c7d71183b9ef859449ab94a390d8a9fbcd3a73d5/dev-docs).

A skeleton project has been created with a few documents in it. The rest of the content needs adding from the wiki. This issue attempts to identify the pages that we want to keep and have transferred to the new `dev-docs` project. The last section gives a list of pages that should *not* be migrated as they are not considered useful any longer.

I've attempted to group pages into various sections  to try and identify pages that are either duplicates or should be considered for merging and updating.

This issue is about migrating/creating the new pages in the `dev-docs` project and crucially **not** yet updating the main `index.rst` page. This is so that multiple developers can work on this simultaneously without having merge conflicts in the final PRs. A new issue will be created to craft a sensible index page once more of this work has been completed.

To help with this effort:

* select a set of pages to migrate (please take related pages as well)
* add your name next to each page
* do the migration:
  * read existing content and decide if it is good in it's current state and 
    * if the page will remain largely unchanged the consider using [pandoc](http://pandoc.org/) to automate the conversion
    * rewrite the page as you see fit 
* when finished open a pull request as usual but use "Refs: ####" rather than "Fixes: ####"
* repeat cycle with next set of pages

### Intro/Getting Started Related

*  [Creating a Mantid developer account](http://www.mantidproject.org/Creating%20a%20Mantid%20developer%20account) - Rename to "Developer Accounts"
*  Amalgamate following pages regarding first-time checkout/build of mantid into single page with sections. This has ben started at http://developer.mantidproject.org/GettingStarted.html.
   *  [Checking out the Mantid Code](http://www.mantidproject.org/Checking%20out%20the%20Mantid%20Code)
   *  [Building the Mantid Framework Code](http://www.mantidproject.org/Building%20the%20Mantid%20Framework%20Code)
   *  [Building with CMake](http://www.mantidproject.org/Building%20with%20CMake)
*  [Git Workflow](http://www.mantidproject.org/Git%20Workflow)
   * [Working with git](http://www.mantidproject.org/Working%20with%20git) - should this become a link to an external git cheat sheet on the above page?
*  [Mantid Issue Tracking](http://www.mantidproject.org/Mantid%20Issue%20Tracking)
*  [Coding Standards](http://www.mantidproject.org/Coding%20Standards)
* [The automated build process](http://www.mantidproject.org/The%20automated%20build%20process)

### General Guides

*  [Communication](http://www.mantidproject.org/Communication)
*  [Dependencies](http://www.mantidproject.org/Dependencies)
*  [Handling XML](http://www.mantidproject.org/Handling%20XML)
*  [Useful Tools for Developers](http://www.mantidproject.org/Useful%20Tools%20for%20Developers)
   * [IWYU](http://www.mantidproject.org/IWYU)
   * [Profiling With Valgrind](http://www.mantidproject.org/Profiling%20With%20Valgrind) - Maybe make this a general Profiling page with sections for tools on each OS?
   * [Memory Leak Detection](http://www.mantidproject.org/Memory%20Leak%20Detection)
*  [Visual Studio Build Impact](http://www.mantidproject.org/Visual%20Studio%20Build%20Impact)
*  [Debugging Unit Tests](http://www.mantidproject.org/Debugging%20Unit%20Tests)
*  [Doxygen Setup](http://www.mantidproject.org/Doxygen%20Setup) - Use single page to describe setup for all OSes and merge with page below
   * [Unix Console Doxygen Setup](http://www.mantidproject.org/Unix%20Console%20Doxygen%20Setup)
   * [Visual Studio Doxygen Setup](http://www.mantidproject.org/Visual%20Studio%20Doxygen%20Setup)
   * [Eclipse Doxygen Setup](http://www.mantidproject.org/Eclipse%20Doxygen%20Setup)
* [Parallel Mantid using MPI](http://www.mantidproject.org/Parallel%20Mantid%20using%20MPI)
* [Python 3](http://www.mantidproject.org/Python%203)
* [MantidPlot Under Putty](http://www.mantidproject.org/MantidPlot%20Under%20Putty)
* [Python Vs C++ Algorithms](http://www.mantidproject.org/Python%20Vs%20C%2B%2B%20Algorithms)

### GUI Development

* The following pages would be better together:
  * [GUI Design Guidelines](http://www.mantidproject.org/GUI%20Design%20Guidelines)
  * [MVP](http://www.mantidproject.org/MVP)

### Testing
* [Unit Test Good Practice](http://www.mantidproject.org/Unit%20Test%20Good%20Practice)
  * Rework section on test guidance: Private/protected code complex enough to require dedicated testing should be in a separate class/classes
  * Good practice should discourage using files/network and point to helper classes in "Testing Utilities" below
* Split out the data section from the first link and combine it with the "Data Files in Mantid" page.
  * [Data and Helpers for unit tests](http://www.mantidproject.org/Data%20and%20Helpers%20for%20unit%20tests) - Rename to "Testing Utilities"
  * [Data Files in Mantid](http://www.mantidproject.org/Data%20Files%20in%20Mantid) - Rename to "Data Files for Testing"
* [Running the unit tests](http://www.mantidproject.org/Running%20the%20unit%20tests)
* [Writing Performance Tests](http://www.mantidproject.org/Writing%20Performance%20Tests)
* [System Tests](http://www.mantidproject.org/System%20Tests)

### Documentation-related
[... 32 lines cut ...]
[Git Libraries Evaluation](http://www.mantidproject.org/Git%20Libraries%20Evaluation)
[Compiler Warnings](http://www.mantidproject.org/Compiler%20Warnings)
[C++11](http://www.mantidproject.org/C%2B%2B11)
[Distribution Folder Structure](http://www.mantidproject.org/Distribution%20Folder%20Structure)
[Upgrading Numpy](http://www.mantidproject.org/Upgrading%20Numpy)
[Upgrading to Visual Studio 2012](http://www.mantidproject.org/Upgrading%20to%20Visual%20Studio%202012)
[Different Size Types](http://www.mantidproject.org/Different%20Size%20Types)
[Qt designer with custom plugins](http://www.mantidproject.org/Qt%20designer%20with%20custom%20plugins)
[NSIS CPACK Customisations](http://www.mantidproject.org/NSIS%20CPACK%20Customisations)
[Framework Code Directory Structure](http://www.mantidproject.org/Framework%20Code%20Directory%20Structure)
[Debugging Python](http://www.mantidproject.org/Debugging%20Python)
[Development Tools and Frameworks](http://www.mantidproject.org/Development%20Tools%20and%20Frameworks)
[Geometry Development Introduction](http://www.mantidproject.org/Geometry%20Development%20Introduction)
[Python GUI Control in MantidPlot](http://www.mantidproject.org/Python%20GUI%20Control%20in%20MantidPlot)
[Mantid Prerequisites](http://www.mantidproject.org/Mantid%20Prerequisites)
[How to rollback a commit](http://www.mantidproject.org/How%20to%20rollback%20a%20commit)
[How to fix CppCheck Errors](http://www.mantidproject.org/How%20to%20fix%20CppCheck%20Errors)
[How to fix Doxygen Errors](http://www.mantidproject.org/How%20to%20fix%20Doxygen%20Errors)
[Integrate non Mantid code projects with Mantid](http://www.mantidproject.org/Integrate%20non%20Mantid%20code%20projects%20with%20Mantid)
[Mantid Package Rules](http://www.mantidproject.org/Mantid%20Package%20Rules)
[Packaging ParaView Plugins On OS X](http://www.mantidproject.org/Packaging%20ParaView%20Plugins%20On%20OS%20X)
[PythonInterface Return Value Policies](http://www.mantidproject.org/PythonInterface%20Return%20Value%20Policies)
[Static Analysis](http://www.mantidproject.org/Static%20Analysis)
[Stress Tests](http://www.mantidproject.org/Stress%20Tests)
[Working on Core Functionality](http://www.mantidproject.org/Working%20on%20Core%20Functionality)

reimundILL, 2018-04-03:
@martyngigg MantidPlot under Putty is still under construction. The current pages do not seem to be very up to date. I got a new Windows notebook and try to figure out concurrent and best solutions. At the moment I prefer to not install Putty & XMing and not install anything on Windows (using existing Remote Desktop) but software on Linux side only. That may be convenient for most users when logging in on Linux (instrument machines) from Windows (office or home computing).

martyngigg, 2018-04-03:
@reimundILL Thanks for letting me know. Most of the documents are converted now but as this is a single page then I think it can just be added whenever it is ready to go in.

PR #22256 (2018-04-06): Create index page for developer documentation
Description of work.

Creates a landing page for the developer documentation with a first cut at a sensible grouping and order. This can and should evolve over time.

**To test:**

* build the `dev-docs-html` target
* view `build/dev-docs/html/index.html` in a browser

Fixes #22082

**Release Notes** 
*Does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [... cut]
closed 2018-04-06 by peterfpeterson