mantidproject/mantid #22087: Uncaught exception when masking in instrument view

keeeto, 2018-03-14:
### Expected behavior

When attempting to apply a mask to data over part of the ToF range Mantid should not throw an uncaught exception warning. Either the mask should be applied, or a warning should be thrown.

### Actual behavior

Mantid throws an uncaught exception warning.

### Steps to reproduce the behavior

* Open some data eg `INTER00013460` from the training data.
* Open instrument view.
* Reduce ToF so that it does not include the full range.
* In the draw tab, choose any mask shape and draw on the view.
* Click `Apply to Data`

### Platforms affected

Windows
OSX

KarlPalmen, 2018-07-24:
How does one reduce ToF so that it does not include the full range?

KarlPalmen, 2018-07-24:
I've figured out, it is the time range at the bottom of the instrument view window and have reproduced the bug.

PR #22983 (2018-07-25): Fixed exception in instrument view.
Fixed exception caused by masking an instrument without full TOF.

Fixes #22087

**Description of work.**
Caught an unhandled exception caused by masking an instrument with less than the full TOF

**To test:**

See linked issue.

Does this update require release notes?
- [ ] Yes
- [x] No

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [x] Is the code of an acceptable quality?
- [x] Does the code conform to the [coding standards](http://www.mant [... cut]
closed 2018-07-25 by martyngigg