mantidproject/mantid #22093: Incorrect rounding in SumOverlappingTubes

quantumsteve, 2018-03-14:
### Expected behavior

Use `std::lround` and `boost::math::iround` to fix warnings from [misc-incorrect-roundings](http://releases.llvm.org/5.0.1/tools/clang/tools/extra/docs/clang-tidy/checks/misc-incorrect-roundings.html)

### Actual behavior

system test `PowderDiffILLDetScanReductionTest` fails when a naive rounding formula is replaced with `boost::math::iround`. (The naive formula fails when the integer is < 0)

### Steps to reproduce the behavior

1. Apply attached patch
2. Build mantid
3. run system test `PowderDiffILLDetScanReductionTest`

### Platforms affected

all

gvardany, 2018-03-15:
Does the unit test of `SumOverlappinTubes` also fail?

quantumsteve, 2018-03-15:
No, `SumOverlappinTubes` still passes. The issue only occurs when `angleIndex < 0`, which happens in `PowderDiffILLDetScanReductionTest`.

PR #22164 (2018-04-17): D2B detector efficiency calculation
This PR extends the `PowderDiffILLDetEffCorr` algorithm to support the efficiency calculation for D2B instrument, which is a scanning powder diffractometer at the ILL. 

Some satellite algorithms are also fixed/adjusted correspondingly; there were minor bug fixes and extensions.

**To test:**
Use the files 532008 and 532009 from system test data under ILL/D2B directory (set facility to ILL) : 
```
PowderDiffILLDetEffCorr(CalibrationRun='532008,532009', DerivationMethod='GlobalSummedReference2D', OutputWorkspace='calib')
```
The output should be a 2D map of d [... cut]
closed 2018-04-17 by martyngigg