mantidproject/mantid #22098: Clean up SASView save algorithms

Mantid-Matthew, 2018-03-15:
The algorithms used to save out .xml files for use by SASView are more complicated that they need to be. There are currently two versions which write out different versions of the SASView format. Look into whether the old one can be removed and tidy up the new one.

gemmaguest, 2020-07-30:
@smk78 We're not sure why this was on your priorities list because it seems quite low priority?

Is it the tidy up or the removal that is important (or both)?

smk78, 2020-07-30:
Certainly we should remove SaveCanSAS1D v1 (or at least make it really obvious it should not be routinely used).

Does LoadCanSAS v2 read output from SaveCanSAS1D v1? If it does, then LoadCanSAS1D v1 can be dispensed with. If not, I would keep it for backward compatibility.

The tidying up aspect cuts also across the elements in https://github.com/mantidproject/mantid/issues/28458, https://github.com/mantidproject/mantid/issues/28847 and https://github.com/mantidproject/mantid/issues/28998 .

closed 2021-03-27 by stale[bot]