mantidproject/mantid #22123: DetectorEfficiencyCor-v1 does not multiply data by ki/kf

gmonet, 2018-03-16:
The doc for the algorithm "[DetectorEfficiencyCor-v1](https://github.com/mantidproject/mantid/blob/master/docs/source/algorithms/DetectorEfficiencyCor-v1.rst)" says that data are multiplied by ki/kf at the end of the correction but there is no trace of this operation in the c++ source code. 
Moreover, some interfaces (ConvertToEnergy (Direct)) actually do CorrectKiKf operation after DetectorEfficiencyCor...

AndreiSavici, 2018-03-16:
The data is multiplied with 1/k_f in the code. Since k_i is a constant, it is the same operation

gmonet, 2018-03-16:
Well... In this case, i don't understand why there is a correction by k_i/k_f in the [DirectEnergyConversion ](https://github.com/mantidproject/mantid/blob/master/scripts/Inelastic/Direct/DirectEnergyConversion.py) script ([line 1798](https://github.com/mantidproject/mantid/blob/master/scripts/Inelastic/Direct/DirectEnergyConversion.py#L1798)) although there is a correction of the efficiency of the detector [line 1731](https://github.com/mantidproject/mantid/blob/master/scripts/Inelastic/Direct/DirectEnergyConversion.py#L1731) :/

gmonet, 2018-03-19:
In instrument parameters :
https://github.com/mantidproject/mantid/blob/546b2eccbaf236fd217ccbd21573f88aa0272103/instrument/MARI_Parameters.xml#L390-L398
Did i miss something ? :(

gmonet, 2018-05-15:
My bad, we actually need to correct data by the detector efficiency AND to multiply by ki/kf.
The documentation of DetectorEfficiencyCor-v1 is, maybe, misleading.

closed 2018-05-15 by gmonet