mantidproject/mantid #22179: Indirect - ConvFit - Mantid Crashes on Ubuntu and Redhat when fit is run with more than a single fit function

brandonhewer, 2018-03-22:
In the Interfaces > Indirect > Data Analysis > ConvFit interface, running the fitting routine with a composite function (e.g. One Lorentzian and a Delta Function) results in a crash on Ubuntu and Red Hat systems.

The exception ```Function index (2) out of range (2)``` is thrown within CompositeFunction.

This bug is not repeatable on Windows (7 or 10) or Fedora.

NickDraper, 2018-07-05:
Is this genuinely high priority (i.e. actually stopping someone doing work? with no work around).

If it is then add a comment stating who is impacted and how, otherwise please remove the high priority label.

brandonhewer, 2018-07-19:
This is no longer an issue -- it was working on all tested versions of Ubuntu but now Ubuntu 14.04 is also no longer supported.

closed 2018-07-19 by brandonhewer