mantidproject/mantid #22212: Upgrade LoadILLSANS to incorporate the changes of the nexus structure.

gvardany, 2018-03-27:
ILL SANS loader has to be upgraded to match the new requirements.

reimundILL, 2018-05-17:
LoadILLSANS-[Error] Error in execution of algorithm LoadILLSANS:
May need to be addressed:
Load should not execute if file is not ill sans, but:
```Python
LoadILLSANS-[Error] Unable to open File: in /home/cs/reimund/Documents/
MantidUI-[Notice] MantidPlot is shutting down...
NXRoot: Error loading /home/cs/reimund/Documents/JS_minus_H2O50D2O50.savNXRoot: Error loading /home/cs/reimund/Documents/HDF5: infinite loop closing library
      D,T,F,FD,P,FD,P,FD,P,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E,E
```

D22 current Nexus files:
```Python
Error in execution of algorithm LoadILLSANS:
SpectrumInfo: No detectors for this workspace index. search object
```

reimundILL, 2018-05-18:
Conversion from size_t to double line 307
HistogramData::BinEdges histoBinEdges(vectorSize, HistogramData::LinearGenerator(0.0, vectorSize));

PR #22517 (2018-06-06): LoadILLSANS
This upgrades the SANS loader to support the modern format for D11, D22 and D33 instruments at ILL.

**To test:**

Load the 4 nexus files you find in unit test data under ILL/D11, ILL/D22, ILL/D33 (set facility to ILL first).
Check that the output has the instrument attached (right click->show instrument).
The rest is captured by the unit tests.

Fixes #22212 . 

Also merge PR [#109](https://github.com/mantidproject/mantidgeometry/pull/109) in geometry.

Does this update require release notes?
- [x] Yes
- [ ] No

---

#### Reviewer ####

Please comment on the following ([full descr [... cut]
closed 2018-06-06 by SimonHeybrock