mantidproject/mantid #22241: MantidPlot segfaults on exit with sip 4.19.8

rosswhitfield, 2018-03-29:
Downgrading to `4.19.7` fixes the problem. The only significant change from `4.19.7` to `4.19.8` was [this](https://www.riverbankcomputing.com/hg/sip/rev/71bfa703c4ee).

The same crash was [observed with krita](https://bugs.kde.org/show_bug.cgi?id=391992) and discussed on the [PyQt mailing list](https://www.riverbankcomputing.com/pipermail/pyqt/2018-March/040140.html).

```
FrameworkManager-[Notice] Welcome to Mantid 3.12.20180328.1335
FrameworkManager-[Notice] Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid
Python-[Warning] Failed to load plugin /home/rwp/mantid/Framework/PythonInterface/plugins/algorithms/WorkflowAlgorithms/WANDPowderReduction.py. Error: invalid syntax (WANDPowderReduction.py, line 76)
MantidUI-[Notice] MantidPlot is shutting down...

********* SEGMENTATION FAULT *********
 /home/rwp/mantid/Release/py3/bin/libMantidAPI.so(+0x193309) [0x7f25029b1309]
 /home/rwp/mantid/Release/py3/bin/libMantidAPI.so(+0x19351f) [0x7f25029b151f]
 /usr/lib/libc.so.6(+0x348e0) [0x7f24f94278e0]
 /usr/lib/libpthread.so.0(sem_wait+0x9) [0x7f24f97b9e39]
 /usr/lib/libpython3.6m.so.1.0(PyThread_acquire_lock_timed+0xe6) [0x7f24fc5d0f26]
 /usr/lib/libpython3.6m.so.1.0(+0x1ec0cd) [0x7f24fc6c70cd]
 /usr/lib/libpython3.6m.so.1.0(+0x82d56) [0x7f24fc55dd56]
 /usr/lib/python3.6/site-packages/sip.so(+0x13345) [0x7f24de2eb345]
 /usr/lib/python3.6/site-packages/sip.so(+0x134d3) [0x7f24de2eb4d3]
 /usr/lib/python3.6/site-packages/PyQt4/QtGui.so(+0x1c0a60) [0x7f24dd5cca60]
 /usr/lib/python3.6/site-packages/PyQt4/QtGui.so(+0x1c0a89) [0x7f24dd5cca89]
 /usr/lib/libQtCore.so.4(_ZN14QObjectPrivate14deleteChildrenEv+0x84) [0x7f24fd6b31c4]
 /usr/lib/libQtGui.so.4(_ZN7QWidgetD2Ev+0x34f) [0x7f24fde78ddf]
 /usr/lib/libQtGui.so.4(_ZN11QDockWidgetD0Ev+0xa) [0x7f24fe21d78a]
 /usr/lib/libQtCore.so.4(_ZN14QObjectPrivate14deleteChildrenEv+0x84) [0x7f24fd6b31c4]
 /usr/lib/libQtGui.so.4(_ZN7QWidgetD2Ev+0x34f) [0x7f24fde78ddf]
 /home/rwp/mantid/Release/py3/bin/MantidPlot(+0x119c89) [0x5591c5d00c89]
 /usr/lib/libQtCore.so.4(_ZN7QObject5eventEP6QEvent+0x201) [0x7f24fd6b4cb1]
 /usr/lib/libQtGui.so.4(_ZN7QWidget5eventEP6QEvent+0xccd) [0x7f24fde79d7d]
 /usr/lib/libQtGui.so.4(_ZN11QMainWindow5eventEP6QEvent+0x11c) [0x7f24fe24e5cc]
 /home/rwp/mantid/Release/py3/bin/MantidPlot(+0x179962) [0x5591c5d60962]
 /usr/lib/libQtGui.so.4(_ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent+0x94) [0x7f24fde251f4]
 /usr/lib/libQtGui.so.4(_ZN12QApplication6notifyEP7QObjectP6QEvent+0x22d) [0x7f24fde2be4d]
 /home/rwp/mantid/Release/py3/bin/MantidPlot(+0x48acf3) [0x5591c6071cf3]
 /usr/lib/libQtCore.so.4(_ZN16QCoreApplication14notifyInternalEP7QObjectP6QEvent+0x8e) [0x7f24fd69aeae]
 /usr/lib/libQtCore.so.4(_ZN23QCoreApplicationPrivate16sendPostedEventsEP7QObjectiP11QThreadData+0x303) [0x7f24fd69db43]
 /usr/lib/libQtCore.so.4(_ZN16QCoreApplication4execEv+0xd8) [0x7f24fd69ee28]
 /home/rwp/mantid/Release/py3/bin/MantidPlot(+0xb9f9f) [0x5591c5ca0f9f]
 /usr/lib/libc.so.6(__libc_start_main+0xea) [0x7f24f9413f4a]
 /home/rwp/mantid/Release/py3/bin/MantidPlot(+0xbabfa) [0x5591c5ca1bfa]
```
Any thoughts @martyngigg ?

martyngigg, 2018-04-03:
@rosswhitfield Off the top of my head, I can see a `QDockWidget` in the stack trace and the only one we expose to Python is the one for the IPython console so I wonder if it is something to do with that? Perhaps try commenting out lines [576-584 in ApplicationWindow](https://github.com/mantidproject/mantid/blob/master/MantidPlot/src/ApplicationWindow.cpp#L576) to see if it prevents the crash?

rosswhitfield, 2018-04-03:
@martyngigg Yes that does fix it! But it's not obvious to me where the real problem is.

I wonder if someone else can reproduce this, maybe someone on OSX can bump their sip version?

martyngigg, 2018-04-04:
@rosswhitfield One thing that I've just spotted is that we don't actually delete the interpreter dock, we leave that up to the Qt parent object. As this dock contains a Python object we probably should be destroying that stuff before we shutdown the internal python environment in ApplicationWindow. Can you try adding a `delete m_interpreterDock` to [ApplicationWindow::closeEvent](https://github.com/mantidproject/mantid/blob/master/MantidPlot/src/ApplicationWindow.cpp#L9831)?

rosswhitfield, 2018-04-04:
@martyngigg :+1: Yep that fixes it. Thanks! Did you want to PR that and I will test it :grinning:

PR #22251 (2018-04-06): Delete interpreter dock before python shutdown
Description of work.

Manually delete the interpreter dock in MantidPlot before Qt does it. See commit message for more details.

**To test:**

* compile MantidPlot against sip >= 4.19.8
* start and then shutdown MantidPlot => no crash 

Fixes #22241

**Release Notes** 
*Minor changed that does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does th [... cut]
closed 2018-04-06 by quantumsteve